bulletphysics/bullet3 #2410: Got different rendered images with/without eglRendererPlugin

taochenshh, 2019-09-18:
Hi,

I am trying out the eglRendererPlugin. But I got different RGB images when eglRendererPlugin is used and when eglRendererPlugin is not used.

Here is the simple script to recreate the problem:
```python
import pybullet as p
import pybullet_data
import numpy as np
import matplotlib.pyplot as plt
import pkgutil

physicsClient = p.connect(p.DIRECT)  # or p.DIRECT for non-graphical version
# using the eglRendererPlugin (hardware OpenGL acceleration)
egl = pkgutil.get_loader('eglRenderer')
if egl:
    p.loadPlugin(egl.get_filename(), "_eglRendererPlugin")
else:
    p.loadPlugin("eglRendererPlugin")
p.setAdditionalSearchPath(pybullet_data.getDataPath())  # optionally
p.setGravity(0, 0, -10)

planeStartPos = [0, 0, 0]
planeStartOrientation = p.getQuaternionFromEuler([0, 0, 0])
planeId = p.loadURDF("plane.urdf", planeStartPos, planeStartOrientation)

startOri = p.getQuaternionFromEuler([0, 0, 0])
startPos = [0, 0, 1]
urId = p.loadURDF("r2d2.urdf", startPos, startOri)

width = 640
height = 480

fov = 60
aspect = width / float(height)
near = 0.01
far = 10

view_matrix = p.computeViewMatrixFromYawPitchRoll([0, 0, 1], 3, 0, 0,
                                                  0, 2)
projection_matrix = p.computeProjectionMatrixFOV(fov, aspect, near, far)
images = p.getCameraImage(width,
                          height,
                          view_matrix,
                          projection_matrix,
                          shadow=True,
                          renderer=p.ER_BULLET_HARDWARE_OPENGL)
rgb_opengl = np.reshape(images[2], (height, width, 4)) * 1. / 255.

plt.imshow(rgb_opengl)
plt.show()
```

I got the following image when the EGL is not used:

[image]

But I got the image below when EGL is used:
[image]

I am using Python 2.7 and ubuntu 18.04.

taochenshh, 2019-10-22:
Does anyone have the same issue?

erwincoumans, 2019-10-22:
I get a similar image for both Tiny and EGL. Could you update to latest pybullet? I think something related to egl/camera was fixed recently. It may not yet be in pypi, but I think it is.

Tiny: 
[image]

EGL:
[image]

taochenshh, 2019-10-22:
Thanks for your reply. And I tried to install the latest pybullet from source. But I still got different images if I comment out the following lines:
```python
egl = pkgutil.get_loader('eglRenderer')
if egl:
    p.loadPlugin(egl.get_filename(), "_eglRendererPlugin")
else:
    p.loadPlugin("eglRendererPlugin")
```
I tried python 2.7 and python 3.7 on ubuntu 16.04 and ubuntu 18.04 and got the same issue.

erwincoumans, 2019-10-22:
Does the camera rotate if you run the testrender_egl.py, located in bullet/examples/pybullet/examples?

taochenshh, 2019-10-22:
Yes, I just uninstalled pybullet and built it again. Here is the output. Also, when I ran testrender_egl.py, the camera view doesn't change.

```bash
pybullet build time: Sep 23 2019 17:07:23
Loaded EGL 1.4 after reload.
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX 1060/PCIe/SSE2
GL_VERSION=4.6.0 NVIDIA 396.37
GL_SHADING_LANGUAGE_VERSION=4.60 NVIDIA
Version = 4.6.0 NVIDIA 396.37
Vendor = NVIDIA Corporation
Renderer = GeForce GTX 1060/PCIe/SSE2
ven = NVIDIA Corporation
```

erwincoumans, 2019-10-22:
>> pybullet build time: Sep 23 2019 17:07:23
That date is over a month ago. Can you get a fresh clone from 
git clone  https://github.com/bulletphysics/bullet3 and try again?

taochenshh, 2019-10-22:
The repo on my machine was indeed old, sorry for this. Installing the latest PyBullet did solve the issue. Thank you very much!

closed 2019-10-22 by taochenshh