bulletphysics/bullet3 #2411: Frame-Difference regarding Base Position and Orientation

hwiedDFKI, 2019-09-19:
Hi,

there seems to be a difference in frame definition:

When calling `pb.loadURDF(..., basePosition=(x,y,z), baseOrientation=(x,y,z,w))` the base values refer to the root-frame of the URDF.

On the other hand when calling `pb.getBasePositionAndOrientation()` or `pb.resetBasePositionAndOrientation()` the position and orientation values refer to the inertia position of the root link.

As long as the inertia-frame is congruent with the root-frame everything is fine. But e.g. in models used for Gazebo the inertia is placed at the COM of the respective frame.

E.g. for the intertial entry in the root link:
```xml
<link name="Root_Link">
  <inertial>
    <origin xyz="0.06448 3.0e-05 0.13913" rpy="0.0 -0.0 -1.57079"/>
    <mass value="4.802872"/>
    <inertia ixx="0.03308" ixy="3.59994-06" ixz="0.00154" iyy="0.02199" iyz="7.82419e-06" izz="0.03756"/>
  </inertial>
  ...
</link>
```
With a URDF definition like that, this:
```python
initial_pose = {
  pos: (0.0, 0.0, 1.0),
  rot: (0.0, 0.0, 0.0, 1.0)
}
pb.loadURDF(..., basePosition=initial_pose["pos"], baseOrientation=initial_pose["rot"])
# root frame is at initial_pose
pb.getBasePositionAndOrientation(...)
# returns inertia frame pose in world which is != initial_pose
pb.resetBasePositionAndOrientation(..., initial_state["pos"], initial_state["rot"])
# moves the inertial frame to initial_pose 
pb.getBasePositionAndOrientation(...)
# returns initial_pose but the robot root-frame is not at `initial_pose` anymore
```
will cause the robot to be placed in two different poses.

Regarding the solution I would prefer that both relate to the root-frame, but I think in either way it should relate to the same frame.

erwincoumans, 2019-10-04:
Indeed, there is some difference between the URDF/Gazebo convention  (link/joint frame based) 
 and Bullet's own convention (center of mass based). If you want consistency, always use resetBasePositionAndOrientation and ignore loadURDF position. At this stage it is too much hassle to unify the API (always use Center of Mass coordinates, ignore the URDF convention) so please just only use resetBasePositionAndOrientation.

closed 2019-10-04 by erwincoumans

hwiedDFKI, 2019-10-04:
Thanks for your reply. Maybe you should consider to give all those methods/arguments for both baseCom... and baseLink... to make this clear. As in robotics you oftenly are interested in the frame position, or in the CoM of the whole system, but rarely in the CoM of a specific frame.