bulletphysics/bullet3 #2412: Clarify the differences between the supported control modes

diegoferigo, 2019-09-19:
Reading the documentation, it is not entirely clear to me what are the differences of the joint control modes supported in pybullet.

I try to recap them here, please @erwincoumans feel free to jump in to add more details and address holes in my understanding.

Let's assume that in all cases (excluding torque), the reference error is the following:

```
error = kp *  (pos_des - pos) + kd * (vel_des - vel)
```

- **`VELOCITY_CONTROL`**:  This is a pure proportional control for velocity that by default has `kp=0` and `kd=1`.
- **`POSITION_CONTROL`** (alias for `CONTROL_MODE_POSITION_VELOCITY_PD`):  Similar to **`VELOCITY_CONTROL`**. By default it has `kp=1` and `kd=0`. It allows to optionally specify both the target position and target velocity.
- **`TORQUE_CONTROL`**: It should be enabled by disabling the default motorized joints passing a `force=0`. Later calls using this mode will send directly the torque to the joint.
- **`PD_CONTROL`**: Similar to `POSITION` control but it allows specifying not normalized gains.

Questions:
- Is `kp` ignored in `VELOCITY_CONTROL`?
- What is the logic behind the normalized gains in `VELOCITY_CONTROL` and `POSITION_CONTROL`?
- Is `kd=0` by default in `POSITION_CONTROL`?
- Is correct that in `POSITION_CONTROL` if `kd=0` and a velocity reference is specified, it has no effect?
- Is the description of `PD_CONTROL` correct? There's not much information about it

Related issues https://github.com/bulletphysics/bullet3/issues/2152 https://github.com/bulletphysics/bullet3/issues/1026

Steven89Liu, 2019-09-26:
i also have these questions, thanks in advance.

raunaqbhirangi, 2020-06-13:
Any updates on this?

closed 2021-04-26 by erwincoumans