bulletphysics/bullet3 #2415: Visual Studio 2019 Clang 8.0.0

mattparks, 2019-09-21:
I'm trying to compile Bullet3 in Visual Studio 2019 16.2.5 using the clang_cl_x64 toolset, I have bullet compiling just fine using msvc_x64_x64 and msvs_x86. This may be related to https://github.com/bulletphysics/bullet3/issues/2280 but I have a few more errors:
```
  C:\PROGRA~2\MICROS~1\2019\COMMUN~1\VC\Tools\Llvm\80D306~1.0\bin\clang-cl.exe  /nologo -TP -DNO_OPENGL3 -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_WIN64 -I_deps\bullet3-src\src /GR- /Gy /GF -m64 -fdiagnostics-absolute-paths  /DWIN32 /D_WINDOWS /W3  /EHsc /fp:fast /wd4244 /wd4267 /MDd /Zi /Ob0 /Od /RTC1 -D_DEBUG /showIncludes /Fo_deps\bullet3-build\src\BulletSoftBody\CMakeFiles\BulletSoftBody.dir\btSoftBodyConcaveCollisionAlgorithm.obj /Fd_deps\bullet3-build\src\BulletSoftBody\CMakeFiles\BulletSoftBody.dir\BulletSoftBody.pdb -c _deps\bullet3-src\src\BulletSoftBody\btSoftBodyConcaveCollisionAlgorithm.cpp
  In file included from _deps\bullet3-src\src\BulletSoftBody\btSoftBodyConcaveCollisionAlgorithm.cpp:16:
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBodyConcaveCollisionAlgorithm.h:19:
  In file included from _deps\bullet3-src\src\BulletCollision/BroadphaseCollision/btCollisionAlgorithm.h:19:
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(325,16): error : overloaded 'operator+' must have at least one parameter of class or enumeration type
          inline __m128 operator+(const __m128 A, const __m128 B)
                        ^
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(330,16): error : overloaded 'operator-' must have at least one parameter of class or enumeration type
          inline __m128 operator-(const __m128 A, const __m128 B)
                        ^
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(335,16): error : overloaded 'operator*' must have at least one parameter of class or enumeration type
          inline __m128 operator*(const __m128 A, const __m128 B)
                        ^
  In file included from _deps\bullet3-src\src\BulletSoftBody\btSoftBodyConcaveCollisionAlgorithm.cpp:16:
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBodyConcaveCollisionAlgorithm.h:21:
  In file included from _deps\bullet3-src\src\BulletCollision/BroadphaseCollision/btBroadphaseInterface.h:21:
  In file included from _deps\bullet3-src\src\BulletCollision/BroadphaseCollision/btBroadphaseProxy.h:20:
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btVector3.h(305,3): error : no matching literal operator for call to 'operator""__FILE__' with arguments of types 'const char *' and 'unsigned long long', and no matching literal operator template
                  btAssert(!fuzzyZero());
                  ^
  W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(127,51): note: expanded from macro 'btAssert'
                          #define btAssert(x) { if(!(x)){printf("Assert "__FILE__ ":%u (%s)\n", __LINE__, #x);__debugbreak();     }}
                                                                         ^
  In file included from _deps\bullet3-src\src\BulletSoftBody\btSoftBodyConcaveCollisionAlgorithm.cpp:16:
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBodyConcaveCollisionAlgorithm.h:23:
  In file included from _deps\bullet3-src\src\BulletCollision/NarrowPhaseCollision/btPersistentManifold.h:20:
  In file included from _deps\bullet3-src\src\LinearMath/btTransform.h:18:
  In file included from _deps\bullet3-src\src\LinearMath/btMatrix3x3.h:19:
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btQuaternion.h(111,3): error : no matching literal operator for call to 'operator""__FILE__' with arguments of types 'const char *' and 'unsigned long long', and no matching literal operator template
                  btAssert(d != btScalar(0.0));
                  ^
  W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(127,51): note: expanded from macro 'btAssert'
                          #define btAssert(x) { if(!(x)){printf("Assert "__FILE__ ":%u (%s)\n", __LINE__, #x);__debugbreak();     }}
                                                                         ^
  In file included from _deps\bullet3-src\src\BulletSoftBody\btSoftBodyConcaveCollisionAlgorithm.cpp:16:
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBodyConcaveCollisionAlgorithm.h:23:
  In file included from _deps\bullet3-src\src\BulletCollision/NarrowPhaseCollision/btPersistentManifold.h:20:
  In file included from _deps\bullet3-src\src\LinearMath/btTransform.h:18:
  In file included from _deps\bullet3-src\src\LinearMath/btMatrix3x3.h:19:
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btQuaternion.h(429,3): error : no matching literal operator for call to 'operator""__FILE__' with arguments of types 'const char *' and 'unsigned long long', and no matching literal operator template
                  btAssert(s != btScalar(0.0));
                  ^
  W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(127,51): note: expanded from macro 'btAssert'
                          #define btAssert(x) { if(!(x)){printf("Assert "__FILE__ ":%u (%s)\n", __LINE__, #x);__debugbreak();     }}
                                                                         ^
  In file included from _deps\bullet3-src\src\BulletSoftBody\btSoftBodyConcaveCollisionAlgorithm.cpp:16:
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBodyConcaveCollisionAlgorithm.h:23:
  In file included from _deps\bullet3-src\src\BulletCollision/NarrowPhaseCollision/btPersistentManifold.h:20:
  In file included from _deps\bullet3-src\src\LinearMath/btTransform.h:18:
  In file included from _deps\bullet3-src\src\LinearMath/btMatrix3x3.h:19:
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btQuaternion.h(437,3): error : no matching literal operator for call to 'operator""__FILE__' with arguments of types 'const char *' and 'unsigned long long', and no matching literal operator template
                  btAssert(s != btScalar(0.0));
                  ^
  W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(127,51): note: expanded from macro 'btAssert'
                          #define btAssert(x) { if(!(x)){printf("Assert "__FILE__ ":%u (%s)\n", __LINE__, #x);__debugbreak();     }}
                                                                         ^
  In file included from _deps\bullet3-src\src\BulletSoftBody\btSoftBodyConcaveCollisionAlgorithm.cpp:16:
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBodyConcaveCollisionAlgorithm.h:23:
  In file included from _deps\bullet3-src\src\BulletCollision/NarrowPhaseCollision/btPersistentManifold.h:20:
  In file included from _deps\bullet3-src\src\LinearMath/btTransform.h:18:
  In file included from _deps\bullet3-src\src\LinearMath/btMatrix3x3.h:19:
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btQuaternion.h(451,3): error : no matching literal operator for call to 'operator""__FILE__' with arguments of types 'const char *' and 'unsigned long long', and no matching literal operator template
                  btAssert(s != btScalar(0.0));
                  ^
  W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(127,51): note: expanded from macro 'btAssert'
                          #define btAssert(x) { if(!(x)){printf("Assert "__FILE__ ":%u (%s)\n", __LINE__, #x);__debugbreak();     }}
                                                                         ^
  In file included from _deps\bullet3-src\src\BulletSoftBody\btSoftBodyConcaveCollisionAlgorithm.cpp:16:
[... 122 lines cut ...]
                  btAssert(!fuzzyZero());
                  ^
  W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btScalar.h(127,51): note: expanded from macro 'btAssert'
                          #define btAssert(x) { if(!(x)){printf("Assert "__FILE__ ":%u (%s)\n", __LINE__, #x);__debugbreak();     }}
                                                                         ^
  In file included from _deps\bullet3-src\src\BulletSoftBody\btSoftBody.cpp:17:
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBodyInternals.h:20:
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBody.h:21:
[... cut ...]
  In file included from _deps\bullet3-src\src\BulletSoftBody/btSoftBody.h:21:
  In file included from _deps\bullet3-src\src\LinearMath/btTransform.h:18:
  In file included from _deps\bullet3-src\src\LinearMath/btMatrix3x3.h:19:
W:\Acid\Acid\Build\x64-Clang-Debug\_deps\bullet3-src\src\LinearMath\btQuaternion.h(429,3): error : no matching literal operator for call to 'operator""__FILE__' with arguments of types 'const char *' and 'unsigned long long', and no matching literal operator template
                  btAssert(s != btScalar(0.0));

```

erwincoumans, 2019-12-03:
Yes, we need to fix the btAssert in Bullet/src/LinearMath/btScalar.h, using #ifdef __clang__ for example.

erwincoumans, 2019-12-18:
Should be fixed now, since this commit:
https://github.com/bulletphysics/bullet3/pull/2537
Thanks for the report!

closed 2019-12-18 by erwincoumans