bulletphysics/bullet3 #2418: Error building on XCode11

spencerkohan, 2019-09-24:
I have a project which was previously linking against bullet 2.88 using XCode 10 (as installed by homebrew) and everything was working perfectly.  Now since upgrading to XCode11 I have an error on line 335 in btVector3.h:

    y = bt_splat_ps(y, 0x80);

    error: Argument value 10880 is outside the valid range [0, 255]

It appears to be the macro invocation on `bt_splat_ps`.

The compiler being used is `clang`, with language dialect `C++17`.

I am not aware what has changed with regard to XCode which could be causing this issue.

wduminy, 2019-10-27:
@spencerkohan I can suggest a  work-around unit the problem is investigated further.

Add this snippet to `bullet/src/LinearMath/btVector3.h`.
Just below `#define BT_VECTOR3_H` 
```cpp
#ifdef __clang__
#pragma clang diagnostic ignored "-Wargument-outside-range"
#endif
```

Bo98, 2019-11-13:
@spencerkohan Just to add here, since I noticed you mentioned Homebrew, I've applied commit https://github.com/bulletphysics/bullet3/commit/7638b7c5a659dceb4e580ae87d4d60b00847ef94 onto the 2.88 source base and an update is now available on Homebrew. If you install the latest from Homebrew it should work now.

erwincoumans, 2019-11-16:
should be fixed now.

closed 2019-11-16 by erwincoumans

collenjones, 2020-01-16:
@erwincoumans FYI the fix does not appear to have solved this issue but @wduminy's suggestion works