bulletphysics/bullet3 #2427: Jacobian Transpose Control

Alescontrela, 2019-09-30:
I'm attempting to apply jacobian transpose control to compute stance leg torques for the Laikago robot. <br/>
[image]

I have calculated F (the contact forces for that leg's foot) and would like to compute the torques required to achieve the calculated foot contact force. I first obtain the jacobian via:

```python
JT, _ = self._env.p.calculateJacobian(sim_id, foot_link_index,
                                             zero_pos, actuated_dof_positions,
                                             zero_vel,
                                             zero_acc)
```

Then I slice out the columns corresponding to leg i's joints via:

```python
JT_leg = np.array(JT)[:, [haa_ind, hfe_ind, kfe_ind]]
```

where `haa_ind`, `hfe_ind`, and `kfe_ind` are the indices for the hip abduction/adduction, hip flexion/extension, and knee flexion/extension joints, respectively. Finally, I compute the leg torques as follows:

```python
leg_torques = np.matmul(J.T, target_leg_force)
```
The resulting behavior of the leg torques does not match what I expect given the desired foot contact forces, so I suspect i'm doing something wrong. After some debugging, I have come across some suspicious behavior. First, on some iterations the sliced jacobian columns for a particular leg result in a zero matrix:

```python
LEGIDX: 3
Jacobian:
 >> [[0. 0. 0.]
     [0. 0. 0.]
     [0. 0. 0.]]
```

Which seems strange... does it make sense for the haa,hfe,kfe columns in the Jacobian for a leg's foot to contain all zeros? 

Additionally, the controller i'm implementing alternates between position control (for swing legs) and torque control (for stance legs). With this in mind, I apply position control as follows:

```python
self.p.setJointMotorControl2(self.sim_id, jointIndex=joint,
                    controlMode=p.POSITION_CONTROL, force=self.max_force,
                    targetPosition=desired_joint_angle)
```

and torque control:

```python
self.p.setJointMotorControl2(self.sim_id, jointIndex=joint,
                    controlMode=self.p.TORQUE_CONTROL, force=desired_joint_torque)
```

Might it be possible that rapid alternation between position and torque control results in strange behavior?

Alescontrela, 2019-10-01:
Never mind. I figured out that the Jacobian is computed in the base frame. Marking issue as resolved.

closed 2019-10-01 by Alescontrela

erwincoumans, 2019-10-02:
Thanks for reporting back, I didn't make time to reply. If would be good to have some better example using control etc. We have some quadruped control (Raibert controller) using the Jacobians, but it isn't open source yet.

vk4151, 2021-02-08:
> Never mind. I figured out that the Jacobian is computed in the base frame. Marking issue as resolved.

Hello Alescontrela, 

Would you please explain me what do you mean by computed in the base frame?

I am trying to implement cartesian impedance controller for KUKA lwr robot. I would want to convert the cartesian forces to Joint torques I tried converting by multiplying with Jacobian Transpose. However, I observed the unexpected behaviour of the robot. 

Would any one please help me to resolve this issue  that the best way to convert Cartesian forces and torques to joint torques using Jacobian obtained from pybullet.calculateJacobian ?