bulletphysics/bullet3 #2434: module 'baselines.deepq' has no attribute 'load'

GoogleMike, 2019-10-09:
Hi, I am new to pybullet and baselines, so sorry, if I am asking a unnecessary question.
When I run the **enjoy_kuka_grasping.py**, some errors happens.
It shows that ###  **module 'baselines.deepq' has no attribute 'load'**.
Is there someone tell me how to fix it?

leeivan1007, 2019-10-16:
@GoogleMike 
it can add the code with:
``` python
def callback(lcl, glb):
    # stop training if reward exceeds 199
    total = sum(lcl['episode_rewards'][-101:-1]) / 100
    totalt = lcl['t']
    #print("totalt")
    #print(totalt)
    is_solved = totalt > 2000 and total >= 10
    return is_solved
load_path = 'kuka_model.pkl'
act = deepq.learn(env, network='mlp',lr=1e-3,total_timesteps=0,
buffer_size=50000,print_freq=10,load_path=load_path,callback=callback)
```
it is not a good solution, but can save this problem.

GoogleMike, 2019-10-16:
> @GoogleMike
> it can add the code with:
>
> ```python
> def callback(lcl, glb):
>     # stop training if reward exceeds 199
>     total = sum(lcl['episode_rewards'][-101:-1]) / 100
>     totalt = lcl['t'] [...]
Hi, I test the code again. It seems that no error about baselines.
Now, it shows that:
> No file named 'kuka_model.pkl'.
It means that I should run **_train_kuka_grasping.py_**, then run  **_enjoy_kuka_grasping.py_**?
I don't know well about that.

leeivan1007, 2019-10-17:
yes, you are right!
maybe you can modify total_timesteps to 10, make sure this can run first.
but it look like will not be converge #1386 
I am trying the [solution](https://github.com/bulletphysics/bullet3/issues/1386#issuecomment-467512724)
maybe we can change the wechat ID,if I have some new progress, I will share it with you.

GoogleMike, 2019-10-17:
> yes, you are right!
> maybe you can modify total_timesteps to 10, make sure this can run first.
> but it look like will not be converge #1386
> I am trying the [...]

My wechat ID is ran1538464

closed 2021-04-26 by erwincoumans