bulletphysics/bullet3 #2437: Unity3D crashed with pybullet (Ubuntu 18.04)

EnricoGiordano1992, 2019-10-15:
I'm trying to use pybullet on Unity3D using your wrapper, but Unity3D crash with this error:

```
Plugins: Couldn't open Assets/Plugins/pybullet.so, error: Assets/Plugins/pybullet.so: undefined symbol: PyCapsule_Type
unity3d() [0x19882c4]
unity3d() [0xa03817]
unity3d() [0x55b42d]
unity3d() [0xa88c54]
unity3d() [0xa8a448]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(mono_lookup_pinvoke_call+0x2ec) [0x7f94f6e1d90d]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(+0x1a9bd1) [0x7f94f6e30bd1]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(+0x6ced9) [0x7f94f6cf3ed9]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(+0x54499) [0x7f94f6cdb499]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(+0x55f11) [0x7f94f6cdcf11]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(+0x59c09) [0x7f94f6ce0c09]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(+0x5a5a3) [0x7f94f6ce15a3]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(+0x1c534e) [0x7f94f6e4c34e]
/opt/Unity3D/Editor/Data/MonoBleedingEdge/MonoEmbedRuntime/libmonobdwgc-2.0.so(mono_runtime_invoke+0x1c) [0x7f94f6e4d1fb]
unity3d() [0xe1010d]
unity3d() [0xec1f44]
unity3d() [0xe7b959]
unity3d() [0xe7bd6f]
unity3d() [0xe7c878]
unity3d() [0x79d4cb]
unity3d() [0xaff2fb]
unity3d() [0xa8d841]
unity3d() [0xa8d927]
unity3d() [0xa8e5ec]
unity3d() [0x19caefc]
unity3d() [0x19cbeb2]
unity3d() [0x19cc58b]
unity3d() [0x19c8d51]
unity3d() [0x1b2b38c]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4cd02) [0x7f955e4bcd02]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x154) [0x7f955e4bc284]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4c64f) [0x7f955e4bc64f]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_loop_run+0xc1) [0x7f955e4bc961]
/usr/lib/x86_64-linux-gnu/libgtk-3.so.0(gtk_main+0x84) [0x7f955f6cba24]
unity3d() [0x4f29e2]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe6) [0x7f955544bb96]
unity3d() [0x535028]
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Misc/Plugins.cpp Line: 236)

Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libpybullet
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libpybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/pybullet
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libpybullet
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libpybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libpybullet
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
DllNotFoundException: pybullet
  at (wrapper managed-to-native) NativeMethods.b3ConnectSharedMemory(int)
  at NewBehaviourScript.Start () [0x00006] in /home/mklab/Unity/test2/Assets/Scenes/examples/NewBehaviourScript.cs:175 
 
(Filename: Assets/Scenes/examples/NewBehaviourScript.cs Line: 175)

Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Fallback handler could not load library /opt/Unity3D/Editor/Data/Mono/lib/libAssets/Plugins/pybullet.so
Receiving unhandled NULL exception
Launching bug reporter
```

I have build pybullet using build_cmake_pybullet_double.sh and I'm able to use pybullet on python3.6 without problems, so I think that there is something strange between Unity3D and pybullet.

In addition, I have modified NativeMethods.cs at line 66:

```
    public const string B3_SHARED_API = "__attribute__((visibility(\"default\")))";
    //public const string B3_SHARED_API = "__declspec(dllexport)";
```

and at line 1421:

```
    //const string dllName = "pybullet_vs2010_x64_release.dll";
    const string dllName = "pybullet";
```

I copied pybullet.so from build_cmake/examples/pybullet into Assets/Plugins directory of Unity3D project.

Any advices?

EnricoGiordano1992, 2019-10-16:
with command ldd -r pybullet.so I get:

```
undefined symbol: PyCapsule_Type	(./pybullet.so)
undefined symbol: _Py_NoneStruct	(./pybullet.so)
undefined symbol: PyExc_AttributeError	(./pybullet.so)
undefined symbol: PyExc_RuntimeError	(./pybullet.so)
undefined symbol: PyExc_ImportError	(./pybullet.so)
undefined symbol: PyTuple_SetItem	(./pybullet.so)
undefined symbol: PyLong_AsLong	(./pybullet.so)
undefined symbol: PyObject_GetAttrString	(./pybullet.so)
undefined symbol: PyDict_SetItem	(./pybullet.so)
undefined symbol: PyModule_AddObject	(./pybullet.so)
undefined symbol: PyUnicode_AsASCIIString	(./pybullet.so)
undefined symbol: PyTuple_New	(./pybullet.so)
undefined symbol: PyArg_ParseTupleAndKeywords	(./pybullet.so)
undefined symbol: PyBytes_FromString	(./pybullet.so)
undefined symbol: PyErr_Format	(./pybullet.so)
undefined symbol: PyFloat_FromDouble	(./pybullet.so)
undefined symbol: PyFloat_AsDouble	(./pybullet.so)
undefined symbol: PyModule_AddIntConstant	(./pybullet.so)
undefined symbol: Py_BuildValue	(./pybullet.so)
undefined symbol: PyImport_ImportModule	(./pybullet.so)
undefined symbol: PyErr_Clear	(./pybullet.so)
undefined symbol: PyDict_New	(./pybullet.so)
undefined symbol: PyErr_SetString	(./pybullet.so)
undefined symbol: PyArg_ParseTuple	(./pybullet.so)
undefined symbol: PyCapsule_GetPointer	(./pybullet.so)
undefined symbol: PySequence_Fast	(./pybullet.so)
undefined symbol: PySequence_GetItem	(./pybullet.so)
undefined symbol: PySequence_Size	(./pybullet.so)
undefined symbol: PyLong_FromLong	(./pybullet.so)
undefined symbol: PyErr_NewException	(./pybullet.so)
undefined symbol: PyModule_Create2	(./pybullet.so)
undefined symbol: Py_AtExit	(./pybullet.so)
undefined symbol: PyErr_Print	(./pybullet.so)
```

EnricoGiordano1992, 2019-10-16:
I have already tried preloading the python so file or adding its path to LOAD_LIBRARY_PATH but nothing happened. Could you explain me more precisely where should I remove pybullet.c from target?

erwincoumans, 2019-10-16:
How did you build pybullet.so? If using cmake, you can just remove it from CMakeLists.txt in bullet/examples/pybullet. If you use python setup.py build, you can remove pybullet.c from setup.py in the root of Bullet.

erwincoumans, 2019-10-22:
this unity pybullet plugin isn't well supported, so you have to figure it out by yourself.

closed 2019-10-22 by erwincoumans