bulletphysics/bullet3 #2442: Non-deterministic bahavior when using position_control and possibly changeConstraint

jyf588, 2019-10-20:
Hi all,

I am using pyBullet to run some simple DRL experiments - thank you for making pyBullet well-documented and shipped with a lot of examples.

I am having an issue with the determinism of pyBullet, which can be important for reproducing RL experiments: it seems that when I am using position_control (or changeConstraint), and when there is contact, sometimes I will see non-determinism. Minimal demo for reproducing this given below:

```
import pybullet as p
import time
import numpy as np

import os
import inspect
currentdir = os.path.dirname(os.path.abspath(inspect.getfile(inspect.currentframe())))

class AllegroHand:
    def __init__(self,
                 base_init_pos=np.array([0., 0, 0.5]),
                 base_init_euler=np.array([1.57, 0, 0])):
        self.baseInitPos = base_init_pos
        self.baseInitOri = base_init_euler

        self.handId = p.loadURDF(os.path.join(currentdir, "assets/allegro_hand_description/allegro_hand_description_right.urdf"),
                                 list(self.baseInitPos), p.getQuaternionFromEuler(list(self.baseInitOri)),
                                 flags=p.URDF_USE_SELF_COLLISION)

        self.savedState = p.saveState()

    def reset(self):
        p.restoreState(self.savedState)

    def get_raw_state_fingers(self):
        joints_state = p.getJointStates(self.handId, range(20))
        joints_state = np.array(joints_state)[:,[0,1]]    # only position, no vel.
        return np.hstack(joints_state.flatten())

    def get_robot_observation(self):
        obs = []
        obs.extend(list(self.get_raw_state_fingers()))
        basePos, baseQuat = p.getBasePositionAndOrientation(self.handId)
        obs.extend(basePos)
        obs.extend(baseQuat)
        return obs

    def get_robot_observation_dim(self):
        return len(self.get_robot_observation())

    def apply_action(self, a):
        # a is not used
        for i in list([5,6,7,8,10,11,12,13,15,16,17,18]):
            p.setJointMotorControl2(self.handId,
                                    jointIndex=i,
                                    controlMode=p.POSITION_CONTROL,
                                    targetPosition=-1.0,
                                    force=300.0)

if __name__ == "__main__":
    physicsClient = p.connect(p.GUI)

    p.setTimeStep(1./240)
    p.setGravity(0, 0, -10)
    p.loadURDF(os.path.join(currentdir, 'assets/plane.urdf'), [0, 0, 0], useFixedBase=1)

    a = AllegroHand()

    for i in range(100):
        a.reset()

        input("press enter to continue")
        print("init", a.get_robot_observation())
        for t in range(500):
            a.apply_action(np.array([np.nan]*22))
            p.stepSimulation()
            time.sleep(1./240.)
        print("final obz", a.get_robot_observation())

    p.disconnect()
```

URDFs used in the code can be downloaded from [here](https://drive.google.com/file/d/1B3AiKrXy7i-wd9uGF8Ump8IEL2DZXMfU/view?usp=sharing)

jyf588, 2019-10-20:
A sample output:
```
press enter to continue
init [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.5, 0.706825181105366, 0.0, 0.0, 0.7073882691671998]
final obz [-0.03496293183241609, -0.11583680659928483, 0.23045265363663797, 0.39648907355113855, -0.16049073737787556, -0.023120696483600822, -0.09308681210526494, 0.3672476272733183, 0.0, 0.0, -0.46952350494861034, -0.09196440003347722, -0.1732771814104369, -0.09039978358461527, -0.17404599262913287, 0.01981621092866076, -0.22649021233881622, -0.31173032125959577, 0.0, 0.0, -0.4710311935846009, -0.035611437166457455, -0.13413768995500341, -0.07464103466422761, -0.18914205872217132, -0.004366532805077237, -0.20791072737649008, -0.2014182525461144, 0.0, 0.0, 0.15714755270111566, -0.03341049956293462, -0.09913777613668193, 0.003652075628614488, -0.18063132428555687, 0.021380798238130567, -0.16620027676483234, -0.11921342184934826, 0.0, 0.0, 0.05703348444533475, 0.09704136805128319, 0.01774218043439315, 0.5276005350859138, 0.4543256948663477, -0.5062525749711718, 0.508855744485926]
press enter to continue
init [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.5, 0.706825181105366, 0.0, 0.0, 0.7073882691671998]
final obz [-0.03142998560095665, 3.438199976947727e-05, 0.228936898630212, 0.05301550091636659, -0.16146731317332857, -0.22969517209719165, -0.08835418603770123, 0.19549457173274754, 0.0, 0.0, -0.4684642789683976, -0.12819820579805719, -0.1724548177587596, -0.002742133038093616, -0.17441746584205234, 0.04211458429314866, -0.22554774260974017, 0.018860874525992208, 0.0, 0.0, -0.47058553100803263, -0.039500636128783545, -0.13344774489369066, 0.013584974613342269, -0.1892728145229466, -0.003751267914577555, -0.2071416017517857, 0.06999078742736752, 0.0, 0.0, 0.15690973211633188, -0.024379489575730388, -0.1000630916670531, 0.011691139885455192, -0.18061927247391715, -0.008770960300902395, -0.1655584376848286, -0.13146469636042762, 0.0, 0.0, 0.05717966824239874, 0.09709474745808125, 0.017822038672508474, 0.5284768860537297, 0.45293669600055547, -0.5073424566511346, 0.5080985750838566]
press enter to continue

```

erwincoumans, 2019-10-22:
I don't have time to look into it now. There is some unit test that checks determinism, with a kuka arm and contact. Could you compare how your setup differs?
See bullet/examples/pybullet/unittests

jyf588, 2019-10-22:
Thanks for the quick response. In https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/unittests/saveRestoreStateTest.py#L22 , if I understand correctly, the target joint position is constant zero, and the maxForce for this position control is zero. But in my case maxForce is non-zero.
Besides, when I was debugging and narrowing down the issue, I did notice that if targetPosition is constant, it is harder to create non-deterministic cases.

erwincoumans, 2019-10-22:
Just a sanity check: if you re-create the world from scratch (loadURDF instead of resetState), is it always deterministic?

jyf588, 2019-10-22:
Yes, if I do resetSimulation and loadURDF, it is always deterministic.

BTW, is the iterative constraint/contact solver of bullet (PGS?) fully deterministic?

erwincoumans, 2019-10-22:
Yes, PyBullet and its constaint solver is fully deterministic as you experienced when using resetSimulation/loadURDF. When using loadState/saveState and one is very careful, it is also deterministic.

jyf588, 2019-10-22:
Thanks. Actually I was not using save/loadState at first - I first saw this problem while I was using resetJointState and resetBase. I then tried save/loadState to see if they would help.

erwincoumans, 2019-10-22:
resetJointState and resetBase is surely never going to be deterministic. The best (but slow) way is using resetSimulation and loadURDF again. Using saveState/restoreState can be more tricky.

Steven89Liu, 2019-10-24:
it is caused by the cached contact points which will make the sequence of the contact point is different between the first time and after restoreState(). it can be fixed if always remove all the contact points by removeContactPoint() in the btPersistentManifold::refreshContactPoints(), but i think it's better to remove these cached points in the suitable place when processRestoreStateCommand() is called.

erwincoumans, 2019-11-17:
@Steven89Liu no, there is a unit test for saveState/restoreState that includes contacts and it is handled ok. restoreState restores the contact points including cached information.

erwincoumans, 2019-11-17:
Add zip file to avoid broken link rot.
[urdfs.zip](https://github.com/bulletphysics/bullet3/files/3856251/urdfs.zip)

Steven89Liu, 2019-11-18:
i don't find any difference between your attached file and the original, could you tell me which file did you change? thanks.
these days i was trying to fix this issue and also find i made a mistake before. so far i find the collision tree is not same between them.

Steven89Liu, 2019-11-18:
i think this time i find the root cause, as the collision tree is different that will cause the sequence of the collision points is different as well.  it can be fixed by sorting the overlapping pairs before narrow collision detect. this is my patch.  but it will decrease the performance.

diff --git a/src/BulletCollision/BroadphaseCollision/btOverlappingPairCache.cpp b/src/BulletCollision/BroadphaseCollision/btOverlappingPairCache.cpp
index 8ce1087..997af79 100644
--- a/src/BulletCollision/BroadphaseCollision/btOverlappingPairCache.cpp
+++ b/src/BulletCollision/BroadphaseCollision/btOverlappingPairCache.cpp
@@ -335,6 +335,7 @@ void btHashedOverlappingPairCache::processAllOverlappingPairs(btOverlapCallback*
        int i;
 
        //      printf("m_overlappingPairArray.size()=%d\n",m_overlappingPairArray.size());
+        sortOverlappingPairs(dispatcher);
        for (i = 0; i < m_overlappingPairArray.size();)
        {
                btBroadphasePair* pair = &m_overlappingPairArray[i];

erwincoumans, 2020-03-01:
Yes, as I mentioned please compare your setup with the unit test in [line 17](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/unittests/saveRestoreStateTest.py#L17).
~~~~~~~~~~~~
 p.setPhysicsEngineParameter(deterministicOverlappingPairs=1)
~~~~~~~~~~~~
That does exactly what you figured out the hard way.

closed 2020-03-01 by erwincoumans