bulletphysics/bullet3 #2444: Constrained z-prismatic point joint state keeps increasing

ajordana, 2019-10-21:
In our setup we have a 1D hopper. We model the hopper using a 1D prismatic joint. See the setup here [1].

When we try to fix the base using something like

```
p.createConstraint(self.robotId, 0, -1, -1, p.JOINT_FIXED, [0, 0, 0], [0, 0, 0.],[0 ,0 , fixed_height])
```

and query the joint state later (while the leg is swinging with zero torque), we observe the `p.getJointState(robot_id, 0)` to increase from the initial fixed_height value=0.4 to values like 4.9. Looking at the visualization, the base stays fixed.

Any idea what is going wrong here? Is the `p.getJointState(...)` returned before taking the constraints into account?

[1]: https://github.com/open-dynamic-robot-initiative/robot_properties_teststand/blob/master/xacro/teststand.urdf.xacro#L14

ajordana, 2019-10-21:
+\cc @jviereck

erwincoumans, 2019-11-28:
Please provide a small self-contained full working pybullet example (without other dependencies).

closed 2021-04-26 by erwincoumans