bulletphysics/bullet3 #2445: Bullet3Collision Shape Definitions

Levi-Armstrong, 2019-10-22:
I currently have an implementation using BulletCollision narrowphase and broadphase for performing collision checking. I am looking to add addition support leveraging Bullet3Collision narrowphase and broadphase. Looking through the API it looks like only the following shapes shown below are supported. Is it correct to assume that for other primitive shapes like cylinder, box, etc I need to use the `SHAPE_CONVEX_HULL`?

```
enum b3ShapeTypes
{
	SHAPE_HEIGHT_FIELD = 1,

	SHAPE_CONVEX_HULL = 3,
	SHAPE_PLANE = 4,
	SHAPE_CONCAVE_TRIMESH = 5,
	SHAPE_COMPOUND_OF_CONVEX_HULLS = 6,
	SHAPE_SPHERE = 7,
	MAX_NUM_SHAPE_TYPES,
};
```

erwincoumans, 2019-10-22:
Yes, Bullet3Collision would use SHAPE_CONVEX_HULL for box and tesselated cylinder.
Unless you are familiar with GPU and OpenCL, I wouldn't use that old OpenCL pipeline. It can be fast but is not well maintained/supported.

closed 2019-10-22 by erwincoumans

Levi-Armstrong, 2019-10-22:
Thank you for the quick response.