bulletphysics/bullet3 #2453: Has the document of pybullet been updated？

Megatron2032, 2019-10-29:
what does URDF_MAINTAIN_LINK_ORDER mean？
Some APIs for deepmimic are not in the document.

erwincoumans, 2019-10-29:
Indeed, doc is missing some api's, please mention them here.
PyBullet link index can be different from their order in the urdf file without that flag:

urdf: parentlink0, link1, link2
pybullet: parentlink0, link2, link1 (without that flag)

Megatron2032, 2019-10-30:
getDifferenceQuaternion
getAxisAngleFromQuaternion

URDF_MAINTAIN_LINK_ORDER

changeDynamics(activationState):
ACTIVATION_STATE_SLEEP
ACTIVATION_STATE_ENABLE_SLEEPING
ACTIVATION_STATE_DISABLE_WAKEUP

setJointMotorControlMultiDof
setJointMotorControlMultiDofArray
STABLE_PD_CONTROL

some missing APIs in deepmimic files.

Megatron2032, 2019-11-01:
some functions used in stablePD：
calculateMassMatrix
calculateInverseDynamics

@erwincoumans

erwincoumans, 2019-11-14:
Add arguments and outputs of getJointStateMultiDof

erwincoumans, 2019-11-17:
Did another pass on updating the quickstart guide. Note that some of those APIs are experimental/obscure and not well documented.

closed 2019-11-17 by erwincoumans