bulletphysics/bullet3 #2463: Pendulum won't swing when using smaller simulation timestep

jyf588, 2019-11-01:
Hi all,

We have this very simple pendulum simulation that swings down passively under gravity. It works fine, though there seems to be some friction (?) that made the pendulum stop at 3.09 instead of \pi.

However, if we *decrease* the timestep from 1/240 to 1/2400, the pendulum will not swing at all. Note that we have tried to turn off all the dampings and frictions that are mentioned in the starter guide.
```

import pybullet as p
import time

physicsClient = p.connect(p.GUI)
p.resetSimulation()
pos = [0,0,0]
orient = p.getQuaternionFromEuler([0,0,0])

robot = p.loadURDF("pendulum.urdf",basePosition=pos,baseOrientation = orient, useFixedBase=1)

p.resetJointState(robot, 0, 0.5)
p.setJointMotorControl2(robot, 0, controlMode=p.TORQUE_CONTROL, force=0.000)

p.changeDynamics(robot, -1, linearDamping=0.0, angularDamping=0.0, contactStiffness=0.0, contactDamping=0.0, lateralFriction=0.0, spinningFriction=0.0, rollingFriction=0.0, jointDamping=0.0)
p.changeDynamics(robot, 0, linearDamping=0.0, angularDamping=0.0, contactStiffness=0.0, contactDamping=0.0, lateralFriction=0.0, spinningFriction=0.0, rollingFriction=0.0, jointDamping=0.0)
# p.changeDynamics(robot, 1, linearDamping=0.0, angularDamping=0.0, contactStiffness=0.0, contactDamping=0.0, lateralFriction=0.0, spinningFriction=0.0, rollingFriction=0.0, jointDamping=0.0)

print(p.getJointInfo(robot,0)[6])	# joint damping = 0.0
print(p.getJointInfo(robot,0)[7])	# joint friction = 0.0

p.setGravity(0, 0, -9.81)
p.setTimeStep(1./240)

for _ in range(100000):
# while (1):
    p.stepSimulation()
    # q = p.getJointState(robot, 0)[0]
    # print(q)
    time.sleep(1 / 240.)
    
p.disconnect()
```

And the model file:
```

<?xml version="1.0" ?>
<robot name="omnipointer">
  <material name="omni/Blue">
    <color rgba="0 0 0.8 1"/>
  </material>
  <material name="omni/Red">
    <color rgba="1 0 0 1"/>
  </material>
  <material name="omni/Green">
    <color rgba="0 1 0 1"/>
  </material>
  <material name="omni/Yellow">
    <color rgba="1 1 0 1"/>
  </material>
  <material name="omni/LightGrey">
    <color rgba="0.6 0.6 0.6 1"/>
  </material>
  <material name="omni/DarkGrey">
    <color rgba="0.4 0.4 0.4 1"/>
  </material>
  <link name="base_link">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0.05715"/>
      <geometry>
        <box size="0.1143 0.0545 0.1143"/>
      </geometry>
      <material name="omni/LightGrey"/>
    </visual>
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0.05715"/>
      <mass value="0.2"/>
      <inertia ixx="0.000267245666667" ixy="0" ixz="0" iyy="0.000435483" iyz="0" izz="0.000267245666667"/>
    </inertial>
  </link>
  <link name="pendulum_link_1">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0.5"/>
      <geometry>
        <box size="0.0402 0.05 1"/>
      </geometry>
      <material name="omni/Red"/>
    </visual>
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0.5"/>
      <mass value="1000.5"/>
      <inertia ixx="83.5416666667" ixy="0" ixz="0" iyy="83.4680033333" iyz="0" izz="0.343003333333"/>
    </inertial>
  </link>
  <joint name="pendulum_joint_1" type="continuous">
    <parent link="base_link"/>
    <child link="pendulum_link_1"/>
    <origin rpy="0 0 0" xyz="0 0.0 0.05715"/>
    <axis xyz="0 1 0"/>
    <dynamics damping="0.0" friction="0.0"/> 
  </joint>
</robot>
```

Steven89Liu, 2019-11-04:
i think it should be caused by the default motor constraint, we can disable this motor constraint with this **p.setJointMotorControl2(robot,0,p.VELOCITY_CONTROL,force=0.000)**.

jyf588, 2019-11-04:
Hi Steven, thanks for the quick reply! And yes, disabling VELOCITY_CONTROL works. This is interesting - why would VELOCITY_CONTROL still be in effect even when I have set Torque Control for the same joint?

Citing Starter Guide here - "Important note: most joints (slider, revolute, continuous) have motors enabled by default that prevent free motion. This is similar to a robot joint with a very high-friction harmonic drive. You should set the joint motor control mode and target settings using pybullet.setJointMotorControl2."

Steven89Liu, 2019-11-05:
this is my understanding, the TORQUE_CONTROL is used to add a torque for some joint while the VELOCITY_CONTROL is used to set the parameters for some joint motor constraint, they should can work together. calling the VELOCITY_CONTROL to disable the default constraint is most likely a workaround, i believe calling the POSITION_CONTROL should also can get the same result.  for more details you can refer to the PhysicsServerCommandProcessor::processSendDesiredStateCommand().

erwincoumans, 2019-11-07:
VELOCITY_CONTROL and POSITION_CONTROL happen through a (persistent) constraint that is satisfied each stepSimulation. By default PyBullet uses VELOCITY_CONTROL with non-zero force to emulate some sort of joint friction. You need to set the VELOCITY_CONTROL or POSITION_CONTROL with zero force to disable it.
TORQUE_CONTROL is just adding a bit of torque, once. It has no influence on the VELOCITY_CONTROL or POSITION_CONTROL constraint.

closed 2019-11-07 by erwincoumans

jyf588, 2019-11-08:
Changed the title of the issue so that people can more easily look it up. It would be great if the starter guide could mention that turning on TORQUE_CONTROL will not turn off the default velocity control.