bulletphysics/bullet3 #2470: Repeated GymEnv instancing/closing causes crash

donthomasitos, 2019-11-07:
Here's the repro code (Ubuntu, Python 3.7):

```
import pybullet_envs
import gym
import gc
for i_episode in range(100):
    env = gym.make('HalfCheetahBulletEnv-v0')
    env.reset()
    for t in range(10):
        action = env.action_space.sample()
        observation, reward, done, info = env.step(action)
    print("Episode finished")
    env.close()
```

This results in "Not connected to physics server" and a program abort. The problem does not occur if these lines are added after each `env.close()`:
```
    gc.collect()
```
Although this is a working workaround, I suggest to make the responsible destruction code being called in `env.close()` instead of waiting for the garbage collector.

erwincoumans, 2019-11-07:
Thanks for the report and repro case!
Indeed, you found a bug in BulletClient (pybullet_utils.bullet_client) that may cause multiple disconnections.
Fixed here: https://github.com/bulletphysics/bullet3/pull/2472
Will be updated in pip pybullet version 2.5.7

closed 2019-11-07 by erwincoumans

donthomasitos, 2019-11-08:
Thanks @erwincoumans for the quick reaction and fix. It now works even without the gc.collect(). You are awesome!

dunkelkaiser, 2022-12-24:
Hi, I have the same problem with physics server in  google colab my evaluation function is source of the problem specifically the env.reset() function  

`def evaluate_policy(policy, eval_episodes=10):
  avg_reward = 0.
  for _ in range(eval_episodes):
    obs = env.reset()
    done = False
    while not done:
      action = policy.select_action(np.array(obs))
      obs, reward, done, _ = env.step(action)
      avg_reward += reward
    env.close()
    gc.collect()
  avg_reward /= eval_episodes
  print ("------------------------------------------------")
  print ("Recompensa promedio en el paso de Evaluación: %f" % (avg_reward))
  print ("------------------------------------------------")
  return avg_reward`
even with  the gc.collect function the mistake persistents

`[/usr/local/lib/python3.8/dist-packages/pybullet_envs/gym_locomotion_envs.py](https://localhost:8080/#) in reset(self)
     28     if (self.stateId >= 0):
     29       #print("restoreState self.stateId:",self.stateId)
---> 30       self._p.restoreState(self.stateId)
     31 
     32     r = MJCFBaseBulletEnv.reset(self)

error: Not connected to physics server.`

I'm using pybullet 3.2.5