bulletphysics/bullet3 #2476: Different colors rendered for the same object in GUI and DIRECT mode

taochenshh, 2019-11-08:
Hi, I have the following script. And I got different images when I use GUI and DIRECT mode.
```python
import matplotlib.pyplot as plt
import numpy as np
import pybullet as p
import pybullet_data

def main():
    p.connect(p.DIRECT)
    p.setAdditionalSearchPath(pybullet_data.getDataPath())
    p.loadURDF('plane.urdf', [0, 0, 0], [0, 0, 0, 1])
    body_id = p.loadURDF('table/table.urdf',
                         basePosition=[1, 0, 0.4],
                         baseOrientation=[0, 0, 0, 1],
                         globalScaling=1)

    focus_pt = [0, 0, 1]  # ([x, y, z])
    width = 640
    height = 480
    view_matrix = p.computeViewMatrixFromYawPitchRoll(focus_pt,
                                                      3,
                                                      0,
                                                      -45,
                                                      0,
                                                      upAxisIndex=2)
    proj_matrix = p.computeProjectionMatrixFOV(60,
                                               width / float(height),
                                               0.01,
                                               10)

    images = p.getCameraImage(width=width,
                              height=height,
                              viewMatrix=view_matrix,
                              projectionMatrix=proj_matrix,
                              shadow=True,
                              flags=p.ER_NO_SEGMENTATION_MASK,
                              renderer=p.ER_BULLET_HARDWARE_OPENGL)
    rgb = np.reshape(images[2],
                     (height, width, 4))[:, :, :3]

    plt.figure()
    plt.imshow(rgb)
    plt.show()

if __name__ == '__main__':
    main()

```

If I use the DIRECT mode, then I will get the following image:
[image]

But If I use the GUI mode, then I will get the following image:
[image]

So the table legs have different colors in DIRECT and GUI modes. Is this possibly a bug? Thanks.

erwincoumans, 2019-11-08:
The URDF file doesn't have a valid color and the renderers deal in different ways with the undefined behavior. You can fix the URDF (like this CL: https://github.com/bulletphysics/bullet3/pull/2477)
or manually iterate over all links and reset the color to a known value:
~~~~~~~~~~~
pybullet.changeVisualShape(objectUid, linkIndex, rgbaColor=[1,1,1,1])
~~~~~~~~~~~

closed 2019-11-08 by erwincoumans

taochenshh, 2019-11-08:
I see. Thanks.