bulletphysics/bullet3 #2480: C Source Code Print statement spamming feed

fbcotter, 2019-11-12:
Similar to the issue raised in #1826, is it possible to disable the `printf` statement `pybullet build time...` in pybullet.c? I have a multiprocess program that uses pybullet and it's clogging the print feed.

erwincoumans, 2019-12-18:
You could edit pybullet.c and modify it for now?

We could add some flag to suppress it (but I'd like to keep this info, so it is more obvious if you use an obsolete version of PyBullet).

luator, 2020-07-01:
I second the request to remove that print.

It causes problems in a setup where I want to run a script using pybullet and parse its output with another script.  It also pops up and interferes with the input line when using ipythons tab-completion in an import statement where pybullet is involved (type `import pybullet.<tab>` to see what I mean).  In general, I think just importing a module should not have any side effects.

I see, that having access to this information can be useful, though.  Would it be an option to not print it automatically but instead provide something like a `get_build_time()` function?

closed 2021-04-26 by erwincoumans