bulletphysics/bullet3 #2483: Velocity loss with CDD after a collision

FlorentMasson, 2019-11-13:
I have a simple example of a ball going through a wall at high speed. Both have restitution = 1.
With CCD off, the ball goes through (as expected).
With CCD on, the ball bounces but loses some of the velocity.
Also, changing the start position for the ball, I get a different velocity after the bounce.
Any idea what's wrong with it? And how I could fix it?

Here's the code

```#include "btBulletDynamicsCommon.h"
#include <iostream>

using namespace std;

int main() {
        btDefaultCollisionConfiguration* ammoCollisionConfiguration = new btDefaultCollisionConfiguration();
        btCollisionDispatcher* ammoDispatcher = new btCollisionDispatcher(ammoCollisionConfiguration);
        btDbvtBroadphase* ammoOverlappingPairCache = new btDbvtBroadphase();
        btSequentialImpulseConstraintSolver* ammoSolver = new btSequentialImpulseConstraintSolver();
        btDiscreteDynamicsWorld* ammoWorld = new btDiscreteDynamicsWorld(ammoDispatcher, ammoOverlappingPairCache, ammoSolver, ammoCollisionConfiguration);
        btVector3 zero(0,0,0);
        ammoWorld->setGravity(zero);

        //wall
        btTriangleMesh* triangleMesh=new btTriangleMesh();
        btVector3 v1(0,0,0);
        btVector3 v2(0,0,0);
        btVector3 v3(0,0,0);
        v1.setX(-10);
        v1.setY(0);
        v1.setZ(-10);
        v2.setX(-10);
        v2.setY(0);
        v2.setZ(10);
        v3.setX(10);
        v3.setY(0);
        v3.setZ(10);
        triangleMesh->addTriangle(v1, v2, v3, true);
        v1.setX(-10);
        v1.setY(0);
        v1.setZ(-10);
        v2.setX(10);
        v2.setY(0);
        v2.setZ(-10);
        v3.setX(10);
        v3.setY(0);
        v3.setZ(10);
        triangleMesh->addTriangle(v1, v2, v3, true);
        btBvhTriangleMeshShape* shape = new btBvhTriangleMeshShape(triangleMesh, true, true);
        btTransform transform;
        transform.setIdentity();
        transform.setOrigin(zero);
        btDefaultMotionState motionState(transform);
        btVector3 localInertia(0, 0, 0);
        btRigidBody::btRigidBodyConstructionInfo rbInfo(0, &motionState, shape, localInertia);
        rbInfo.m_friction = 0;
        rbInfo.m_restitution = 1;
        btRigidBody body(rbInfo);
        ammoWorld->addRigidBody(&body);

        //bullet
        float BULLET_RADIUS = 0.1;
        float BULLET_SPEED = 120.0;
        btTransform transform2;
        transform2.setIdentity();
        transform2.setOrigin(btVector3(0, 21, 0));
        btSphereShape shape2(BULLET_RADIUS);
        btVector3 localInertia2(0, 0, 0);
        shape2.calculateLocalInertia(0.01, localInertia2);
        btDefaultMotionState myMotionState2(transform2);
        btRigidBody::btRigidBodyConstructionInfo rbInfo2(0.01, &myMotionState2, &shape2, localInertia2);
        rbInfo2.m_friction = 0;
        rbInfo2.m_restitution = 1;
        btRigidBody bullet(rbInfo2);
        bullet.setActivationState(4); //DISABLE_DEACTIVATION

        bullet.setAngularFactor(btVector3(0, 0, 0));
        bullet.setLinearVelocity(btVector3(0, -BULLET_SPEED, 0));

        bullet.setCcdMotionThreshold(BULLET_SPEED / 100.0);
        bullet.setCcdSweptSphereRadius(BULLET_RADIUS);

        bullet.setDamping(0, 0);
        ammoWorld->addRigidBody(&bullet);

        btTransform transform3;
        for (int i = 0; i < 50; i++) {
                bullet.getMotionState()->getWorldTransform(transform3);
                btVector3 origin = transform3.getOrigin();
                btVector3 v = bullet.getLinearVelocity();
                cout << "bullet position at tick " << i << " : " << origin.getX() << " " << origin.getY() << " " << origin.getZ() << " velocity " << v.getX() << " " << v.getY() << " " << v.getZ() << "\n";
                ammoWorld->stepSimulation(1 / 60.0, 1);
                int numManifolds = ammoDispatcher->getNumManifolds();
                for (int j = 0; j < numManifolds; j++) {
                        btPersistentManifold* contactManifold = ammoDispatcher->getManifoldByIndexInternal(j);
                        const btRigidBody* obA = btRigidBody::upcast(contactManifold->getBody0());
                        const btRigidBody* obB = btRigidBody::upcast(contactManifold->getBody1());

                        int numContacts = contactManifold->getNumContacts();
                        if (numContacts > 0) {
                                cout << "contacts count " << numContacts << "\n";
                                for (int k = 0; k < numContacts; k++) {
                                        btManifoldPoint c = contactManifold->getContactPoint(k);
                                        btVector3 p1 = c.getPositionWorldOnA();
                                        btVector3 p2 = c.getPositionWorldOnB();
                                        cout << "  distance= " << c.getDistance() << " impulse= " << c.getAppliedImpulse() <<  " p1 " << p1.getX() << " " << p1.getY() << " " << p1.getZ() << " p2 " << p2.getX() << " " << p2.getY() << " " << p2.getZ() << "\n";
                                }
                        }

                }
        }

        return 0;
}
```

Here's the console output:

```bullet position at tick 0 : 0 21 0 velocity 0 -120 0
bullet position at tick 1 : 0 21 0 velocity 0 -120 0
bullet position at tick 2 : 0 19 0 velocity 0 -120 0
bullet position at tick 3 : 0 17 0 velocity 0 -120 0
bullet position at tick 4 : 0 15 0 velocity 0 -120 0
bullet position at tick 5 : 0 13 0 velocity 0 -120 0
bullet position at tick 6 : 0 11 0 velocity 0 -120 0
bullet position at tick 7 : 0 9 0 velocity 0 -120 0
bullet position at tick 8 : 0 7 0 velocity 0 -120 0
bullet position at tick 9 : 0 5 0 velocity 0 -120 0
bullet position at tick 10 : 0 3 0 velocity 0 -120 0
contacts count 1
  distance= 0.9 impulse= 0.66 p1 0 1 0 p2 0 0.1 0
bullet position at tick 11 : 0 1 0 velocity 0 -54 0
contacts count 1
  distance= 0 impulse= 1.08 p1 0 0 0 p2 0 0 0
bullet position at tick 12 : 0 0.1 0 velocity 0 54 0
bullet position at tick 13 : 0 1 0 velocity 0 54 0
bullet position at tick 14 : 0 1.9 0 velocity 0 54 0
bullet position at tick 15 : 0 2.8 0 velocity 0 54 0
bullet position at tick 16 : 0 3.7 0 velocity 0 54 0
bullet position at tick 17 : 0 4.6 0 velocity 0 54 0
bullet position at tick 18 : 0 5.5 0 velocity 0 54 0
bullet position at tick 19 : 0 6.4 0 velocity 0 54 0
bullet position at tick 20 : 0 7.3 0 velocity 0 54 0
bullet position at tick 21 : 0 8.2 0 velocity 0 54 0
bullet position at tick 22 : 0 9.1 0 velocity 0 54 0
bullet position at tick 23 : 0 10 0 velocity 0 54 0
bullet position at tick 24 : 0 10.9 0 velocity 0 54 0
bullet position at tick 25 : 0 11.8 0 velocity 0 54 0
bullet position at tick 26 : 0 12.7 0 velocity 0 54 0
bullet position at tick 27 : 0 13.6 0 velocity 0 54 0
bullet position at tick 28 : 0 14.5 0 velocity 0 54 0
bullet position at tick 29 : 0 15.4 0 velocity 0 54 0
bullet position at tick 30 : 0 16.3 0 velocity 0 54 0
bullet position at tick 31 : 0 17.2 0 velocity 0 54 0
bullet position at tick 32 : 0 18.1 0 velocity 0 54 0
bullet position at tick 33 : 0 19 0 velocity 0 54 0
bullet position at tick 34 : 0 19.9 0 velocity 0 54 0
bullet position at tick 35 : 0 20.8 0 velocity 0 54 0
bullet position at tick 36 : 0 21.7 0 velocity 0 54 0
bullet position at tick 37 : 0 22.6 0 velocity 0 54 0
bullet position at tick 38 : 0 23.5 0 velocity 0 54 0
bullet position at tick 39 : 0 24.4 0 velocity 0 54 0
bullet position at tick 40 : 0 25.3 0 velocity 0 54 0
bullet position at tick 41 : 0 26.2 0 velocity 0 54 0
bullet position at tick 42 : 0 27.1 0 velocity 0 54 0
bullet position at tick 43 : 0 28 0 velocity 0 54 0
bullet position at tick 44 : 0 28.9 0 velocity 0 54 0
bullet position at tick 45 : 0 29.8 0 velocity 0 54 0
bullet position at tick 46 : 0 30.7 0 velocity 0 54 0
bullet position at tick 47 : 0 31.6 0 velocity 0 54 0
bullet position at tick 48 : 0 32.5 0 velocity 0 54 0
bullet position at tick 49 : 0 33.4 0 velocity 0 54 0
```

With a start position of 0,20.5,0 the resulting velocity after bounce is 24

Steven89Liu, 2019-11-19:
thanks for your sharing. this is my understanding.

it is engine's procedure when it detects the collision point by the CCD.
1. create a contact point
2. solve this contact constraint
       1> decrease the the body's velocity to make sure the body can't pass through the other bodies.
            this can explain why we see this log.
                contacts count 1
                      distance= 0.9 impulse= 0.66 p1 0 1 0 p2 0 0.1 0
                bullet position at tick 11 : 0 1 0 velocity 0 -54 0
3. normal contact detecting then the bounced velocity will be 54 that only change the direction because of the m_restitution value of 1.

so the final velocity depends on the distance when the first detect the collision.

i will check if there is any other method to make it more better. thanks.

Steven89Liu, 2019-11-27:
provide a preliminary scheme for reference.
```diff --git a/src/BulletDynamics/ConstraintSolver/btSequentialImpulseConstraintSolver.cpp b/src/BulletDynamics/ConstraintSolver/btSequentialImpulseConstraintSolver.cpp
index e4da468..13bf146 100644
--- a/src/BulletDynamics/ConstraintSolver/btSequentialImpulseConstraintSolver.cpp
+++ b/src/BulletDynamics/ConstraintSolver/btSequentialImpulseConstraintSolver.cpp
@@ -944,9 +944,9 @@ void btSequentialImpulseConstraintSolver::setupContactConstraint(btSolverConstra
 
                if (penetration > 0)
                {
-                       positionalError = 0;
+                       positionalError = -restitution - (penetration - (infoGlobal.m_timeStep - penetration/(-rel_vel)) * restitution) * invTimeStep;
 
-                       velocityError -= penetration * invTimeStep;
+                       //velocityError -= penetration * invTimeStep;
                }
                else
                {
@@ -971,6 +971,8 @@ void btSequentialImpulseConstraintSolver::setupContactConstraint(btSolverConstra
                solverConstraint.m_cfm = cfm * solverConstraint.m_jacDiagABInv;
                solverConstraint.m_lowerLimit = 0;
                solverConstraint.m_upperLimit = 1e10f;
+                if (penetration > 0)
+                   solverConstraint.m_lowerLimit = -1e10f; 
        }
 }
diff --git a/src/BulletDynamics/Dynamics/btDiscreteDynamicsWorld.cpp b/src/BulletDynamics/Dynamics/btDiscreteDynamicsWorld.cpp
index a3c9f42..a4be4ec 100644
--- a/src/BulletDynamics/Dynamics/btDiscreteDynamicsWorld.cpp
+++ b/src/BulletDynamics/Dynamics/btDiscreteDynamicsWorld.cpp
@@ -907,7 +907,7 @@ void btDiscreteDynamicsWorld::createPredictiveContactsInternal(btRigidBody** bod
                                                bool isPredictive = true;
                                                int index = manifold->addManifoldPoint(newPoint, isPredictive);
                                                btManifoldPoint& pt = manifold->getContactPoint(index);
-                                               pt.m_combinedRestitution = 0;
+                                               pt.m_combinedRestitution = gCalculateCombinedRestitutionCallback(body, sweepResults.m_hitCollisionObject);
                                                pt.m_combinedFriction = gCalculateCombinedFrictionCallback(body, sweepResults.m_hitCollisionObject);
                                                pt.m_positionWorldOnA = body->getWorldTransform().getOrigin();
                                                pt.m_positionWorldOnB = worldPointB;

``` ```
Here is the output (i change the step size as this ammoWorld->stepSimulation(1 / 240.0, 1, 1/240.0))
bullet position at tick 38 : 0 2 0 velocity 0 -120 0
bullet position at tick 39 : 0 1.5 0 velocity 0 -120 0
bullet position at tick 40 : 0 1 0 velocity 0 -120 0
bullet position at tick 41 : 0 0.5 0 velocity 0 -120 0
contacts count 1
  distance= 0.4 impulse= 2.4 p1 0 0.5 0 p2 0 0.1 0
bullet position at tick 42 : 0 0.2 0 velocity 0 120 0
bullet position at tick 43 : 0 0.7 0 velocity 0 120 0
bullet position at tick 44 : 0 1.2 0 velocity 0 120 0

FlorentMasson, 2019-11-27:
Thanks, I couldn't reproduce your result with the information provided.
I'm not really keen on lowering the timestep that much, but that may still work in my use case as the "bullet" speed is actually much lower

Steven89Liu, 2019-11-27:
please don't use this patch which maybe cause some other regression issues. if you only want to test, you also need to change your test app like this `ammoWorld->getSolverInfo().m_splitImpulsePenetrationThreshold=10.0;` 
i change the timestep only want to make the test result more easily readable, this following is the result with the old timestep.
```
bullet position at tick 6 : 0 9 0 velocity 0 -120 0
bullet position at tick 7 : 0 7 0 velocity 0 -120 0
bullet position at tick 8 : 0 5 0 velocity 0 -120 0
bullet position at tick 9 : 0 3 0 velocity 0 -120 0
bullet position at tick 10 : 0 1 0 velocity 0 -120 0
contacts count 1
  distance= 0.9 impulse= 2.4 p1 0 1 0 p2 0 0.1 0
bullet position at tick 11 : 0 1.2 0 velocity 0 120 0
bullet position at tick 12 : 0 3.2 0 velocity 0 120 0
bullet position at tick 13 : 0 5.2 0 velocity 0 120 0
bullet position at tick 14 : 0 7.2 0 velocity 0 120 0
bullet position at tick 15 : 0 9.2 0 velocity 0 120 0
bullet position at tick 16 : 0 11.2 0 velocity 0 120 0
bullet position at tick 17 : 0 13.2 0 velocity 0 120 0
bullet position at tick 18 : 0 15.2 0 velocity 0 120 0
bullet position at tick 19 : 0 17.2 0 velocity 0 120 0
bullet position at tick 20 : 0 19.2 0 velocity 0 120 0
```

erwincoumans, 2019-11-28:
At the moment, restitution is not compatible with the CCD feature. For regular contact constraints, the outgoing velocity gets compensated for restitution during solver setup. For CCD contacts this doesn't happen. We need a different approach for CCD contacts to enable restitution.

Steven89Liu, 2019-11-29:
i'm not fully understand this `For CCD contacts this doesn't happen`.  Can we add similiar procedure as the regular contact constraints?
this was my solution, supposing the infoGlobal.m_splitImpulse is true.
1. caculate the contact velocity and the time consumed this stage(ignore other frictions)
        time consumed = penetration/(-rel_vel)
2. get the velocity according to the combined restitution
        restitution = restitutionCurve(rel_vel, cp.m_combinedRestitution)
3. calculate the distance the  body will move after contacting.
        distance = (infoGlobal.m_timeStep - penetration/(-rel_vel)) * restitution
4. the velocity that need to be compensated because of the position error is.
       - (penetration - (infoGlobal.m_timeStep - penetration/(-rel_vel)) * restitution) * invTimeStep
5. additionally we have to compensate the position error as the integration in this timestep
    so the final position error that need to be compensated is,
     `positionalError = -restitution - (penetration - (infoGlobal.m_timeStep - penetration/(-rel_vel)) * restitution) * invTimeStep`

i also thought to split the timestep into two parts but that seems more complicated.

ienapliss, 2020-05-19:
I have a similar issue with CCD, but before my rigidbody hits the ground (static heightfield).

The rigidbody falls from the sky at relatively high velocity but near the ground it suddenly stops in mid air, then continue falling from that 'halted' position losing basically all the energy from the fall.

Is it legit behavior of CCD? Or do I use some wrong parameter?

I set the ccd radius as the length of aabb half extents, and the motion threshold as a small fraction of this (like 0.05 of the radius).

erwincoumans, 2020-05-20:
Unfortunately at the moment the CCD has no restitution and will loose energy. It will require modifications to the implementation.

BluePrintRandom, 2020-05-20:
I roll my own CCD using raycast from points 90 off from linv on a sphere,
I use vector.reflect() and place the object at where it's reflected off the
next frame

I run my CCD after my logic in bge.scene.pre_draw callback

closed 2021-04-26 by erwincoumans