bulletphysics/bullet3 #2488: force/torque sensor under the sole(contact forces)

mohamadmahdavian, 2019-11-16:
Hi 

I am trying to control a humanoid robot in pybullet. But I couldn't find any force/troque sensor to use under the sole to find contact forces between robot feet and the ground. How can I read contact forces between robot and ground?

Thank You

erwincoumans, 2019-11-16:
There are various ways to get this information, for example
1) get all contact points that involve each foot (pybullet.getContactPoints(footId, ...). For each contact point, get the contact force and add them together. The contact force is field [9] of each contact point, according to the [PyBullet Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.cb0co8y2vuvc).

another method is
2)make sure to have a fixed link/joint close to the foot, and enable the force torque sensor at the fixed link/joint (use pybullet.enableJointForceTorqueSensor). After stepSimulation you get access the joint reaction forces/torques from getJointState (field 2 according to the quickstart guide).

closed 2019-11-16 by erwincoumans

BluePrintRandom, 2019-11-16:
a 'ray' sensor could have a real world equivalent in a laser range finder*

tlemarec, 2021-03-30:
I am getting conflicted results from the 2 methods.
For a 5 kg robot : 
- Method 1 : The sum of the normal forces is around 13.6 N.
- Method 2 : The sum of the forces along Fz of each Foot Joints (fixed) is close to 50 N. In this setup the Z axis of those joints in almost perfectly vertical so everything checks out.

erwincoumans, 2021-03-30:
@tlemarec can you provide a reproduction case, with a simple pybullet Python file and urdf file (self-contained)?

tlemarec, 2021-03-31:
@erwincoumans This difference is actually caused by the behavior of substeps (similar to  #1488 I guess). 
I had `numSubSteps = 4`, and observed that the ratio between the two methods was always close to 4.
With `numSubSteps = 1` the two methods give similar results.

Any way to correct this behavior when the number of substep has to be higher than 1 ?