bulletphysics/bullet3 #2491: local joint orientations

asawaswapnil, 2019-11-16:
Here is the modification in URDF from r2d2.urdf

  <joint name="base_to_right_leg" type="continuous">
    <axis xyz="1 1 1"/>
    <parent link="base_link"/>
    <child link="right_leg"/>
    <origin rpy=" 2 0 0" xyz="0.22 0 .25"/>
  </joint>

I am setting a joint angle in urdf = 2 0 0 and axis to any of 1 0 0 or 1 1 1 or anything else. All gives the same result.

1)How can I get the exact **position, rotation, linear velocity, and angular velocity** of joints and links in the **local** frame? I expect something like 2 0 0 for rotation, or some understandable value that doesn't change if the model's global orientation, velocity changes. I want to compare two models by their relative joints, s.t. 2 models walking in the opposite direction but identical walking style, should give the same values. 

2) why does getJointInfo() 2nd last outputs:
(0.0, -1.5707963267948966, -2.0000421244293403)
and getLinkState 5th last output: (-1.570775214158654, 0.42761101912627, -1.570745415938155)
as orientation after converting to Euler? What does this mean and how is it related to 2 0 0?

3)What's the difference between COM vs base vs root in humanoid deep mimic? How to get local and global values of each of them?

erwincoumans, 2019-11-27:
getJointState provides local joint angles and velocities, those don't change when the body base/root moves in world space.

There shouldn't be any contradiction. 
In PyBullet, root == base == CoM of the base.

getBasePositionAndOrientation and resetBasePositionAndOrientation  always refers to the Center of Mass of the base/root.

loadURDF takes some position argument which refers to some URDF specific non-CoM frame that is not supported in PyBullet, please ignore it and just use getBasePositionAndOrientation / resetBasePositionAndOrientation   to avoid confusion.

richardrl, 2019-11-28:
Hey @erwincoumans , how is the COM calculated for mesh inputs? Can you share the code?

For example, I don't understand why this duck mesh looks like this, when I set the base position [0,0,0] (thereby revealing the COM). 
[test.zip](https://github.com/bulletphysics/bullet3/files/3901634/test.zip)

erwincoumans, 2019-11-28:
The CoM is provided in the URDF file, under the 'inertial' element, it is not computed.

erwincoumans, 2019-11-28:
In the case of [duck_vhacd.urdf](https://github.com/bulletphysics/bullet3/blob/master/data/duck_vhacd.urdf), I used VHACD to create an approximate convex decomposition. Also in this case, you can see the 0.02 offset in the inertial field. That way, the duck tumbles as expected.
~~~~~~~~~~~~~~
 <inertial>
      <origin rpy="0 0 0" xyz="0.0 0.02 0.0"/>
       <mass value=".1"/>
       <inertia ixx="1" ixy="0" ixz="0" iyy="1" iyz="0" izz="1"/>
    </inertial>
~~~~~~~~~~~~~~

closed 2019-11-28 by erwincoumans