bulletphysics/bullet3 #2502: getJointInfo returns 0 positions and orientations

asawaswapnil, 2019-11-22:
Similar to #2401 the getjointinfo pybullet call returns 0 orientation for both humanoids. I want to get the local angular rotations for both the humanoids what should I do?

erwincoumans, 2019-11-26:
You need to use getJointStateMultiDof or getJointStatesMultiDof, and it works fine (otherwise deep_mimic wouldn't work. Note that the first joint (root) is a fixed joint.
Here is a modified 'getRewards' that prints the joint state of the chest (spherical/quaternion joint):
~~~~~~~~~~~~~~~~~~~~~~~~
 if useArray:
      jointIndices = range(num_joints)
      simJointStates = self._pybullet_client.getJointStatesMultiDof(self._sim_model, jointIndices)
      kinJointStates = self._pybullet_client.getJointStatesMultiDof(self._kin_model, jointIndices)
      print("simJointStates[1]=",simJointStates[1])
~~~~~~~~~~~~~~~~~~~~~~~~

JOINT_REVOLUTE= 0
JOINT_PRISMATIC,= 1
JOINT_SPHERICAL,= 2
JOINT_FIXED= 4
joint[ 0 ]= (0, b'root', 4, -1, -1, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'root', (0.0, 0.0, 0.0), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0, 1.0), -1)
joint[ 1 ]= (1, b'chest', 2, 7, 6, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'chest', (0.0, 0.0, 0.0), (0.0, 0.166151, 0.0), (0.0, 0.0, 0.0, 1.0), 0)
joint[ 2 ]= (2, b'neck', 2, 11, 9, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'neck', (0.0, 0.0, 0.0), (0.0, 0.10389400000000001, 0.0), (0.0, 0.0, 0.0, 1.0), 1)
joint[ 3 ]= (3, b'right_hip', 2, 15, 12, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'right_hip', (0.0, 0.0, 0.0), (0.0, -0.07, 0.084887), (0.0, 0.0, 0.0, 1.0), 0)
joint[ 4 ]= (4, b'right_knee', 0, 19, 15, 1, 0.0, 0.0, -3.14, 0.0, 1000.0, 0.5, b'right_knee', (0.0, 0.0, 1.0), (0.0, -0.21154599999999998, 0.0), (0.0, 0.0, 0.0, 1.0), 3)
joint[ 5 ]= (5, b'right_ankle', 2, 20, 16, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'right_ankle', (0.0, 0.0, 0.0), (0.0, -0.20987, 0.0), (0.0, 0.0, 0.0, 1.0), 4)
joint[ 6 ]= (6, b'right_shoulder', 2, 24, 19, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'right_shoulder', (0.0, 0.0, 0.0), (-0.02405, 0.1235, 0.18311), (0.0, 0.0, 0.0, 1.0), 1)
joint[ 7 ]= (7, b'right_elbow', 0, 28, 22, 1, 0.0, 0.0, 0.0, 3.14, 1000.0, 0.5, b'right_elbow', (0.0, 0.0, 1.0), (0.0, -0.13478799999999996, 0.0), (0.0, 0.0, 0.0, 1.0), 6)
joint[ 8 ]= (8, b'right_wrist', 4, -1, -1, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'right_wrist', (0.0, 0.0, 0.0), (0.0, -0.138947, 0.0), (0.0, 0.0, 0.0, 1.0), 7)
joint[ 9 ]= (9, b'left_hip', 2, 29, 23, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'left_hip', (0.0, 0.0, 0.0), (0.0, -0.07, -0.084887), (0.0, 0.0, 0.0, 1.0), 0)
joint[ 10 ]= (10, b'left_knee', 0, 33, 26, 1, 0.0, 0.0, -3.14, 0.0, 1000.0, 0.5, b'left_knee', (0.0, 0.0, 1.0), (0.0, -0.21154599999999998, 0.0), (0.0, 0.0, 0.0, 1.0), 9)
joint[ 11 ]= (11, b'left_ankle', 2, 34, 27, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'left_ankle', (0.0, 0.0, 0.0), (0.0, -0.20987, 0.0), (0.0, 0.0, 0.0, 1.0), 10)
joint[ 12 ]= (12, b'left_shoulder', 2, 38, 30, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'left_shoulder', (0.0, 0.0, 0.0), (-0.02405, 0.1235, -0.18311), (0.0, 0.0, 0.0, 1.0), 1)
joint[ 13 ]= (13, b'left_elbow', 0, 42, 33, 1, 0.0, 0.0, 0.0, 3.14, 1000.0, 0.5, b'left_elbow', (0.0, 0.0, 1.0), (0.0, -0.13478799999999996, 0.0), (0.0, 0.0, 0.0, 1.0), 12)
joint[ 14 ]= (14, b'left_wrist', 4, -1, -1, 0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, b'left_wrist', (0.0, 0.0, 0.0), (0.0, -0.138947, 0.0), (0.0, 0.0, 0.0, 1.0), 13)

simJointStates[1]= ((-0.034297708137176565, -0.04078578694456897, 0.02713618025347174, 0.9982103057578101), (-0.1880435685519818, 0.08201809584065678, 0.3598909575762741), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0))
simJointStates[1]= ((-0.03471294384423769, -0.040624979004376414, 0.027810657045443606, 0.9981839459564863), (-0.18409147425074057, 0.07083837927788783, 0.3342893394427152), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0))
simJointStates[1]= ((-0.035116002640777175, -0.0404823549995917, 0.02843989445964863, 0.998157912203948), (-0.17937664331321457, 0.06272353477855708, 0.31206908055257054), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0))
simJointStates[1]= ((-0.035116002640777175, -0.0404823549995917, 0.02843989445964863, 0.998157912203948), (-0.17937664331321457, 0.06272353477855708, 0.31206908055257054), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0))
simJointStates[1]= ((-0.03559296074218341, -0.04037932441809405, 0.029515327750602813, 0.9981139698115231), (-0.20695508789204375, 0.03689358540597154, 0.526861742524436), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0))
simJointStates[1]= ((-0.03606780499070938, -0.04018933176847846, 0.03051882776438898, 0.9980744121592553), (-0.20602962951983195, 0.07984065058110823, 0.4937814951292868), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0))
simJointStates[1]= ((-0.036555347803998064, -0.040027195680062244, 0.03144027813829986, 0.9980345881098035), (-0.21410869322366732, 0.06809661518739629, 0.45424008579775843), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0))
simJointStates[1]= ((-0.03704572187978492, -0.03990361622608748, 0.03228275178161467, 0.997994559023191), (-0.21756787642025555, 0.05105035910110356, 0.4157866991261814), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0))

closed 2019-11-26 by erwincoumans

erwincoumans, 2019-11-29:
By the way, does this work for you?
~~~~~~~~~~~~~~~~~~~~~~~
python3 -m pybullet_envs.deep_mimic.testrl --arg_file run_humanoid3d_walk_args.txt
~~~~~~~~~~~~~~~~~~~~~~~
Attached is a small test that copies the joint positions using getJointStatesMultiDof and copies the joint positions to a test humanoid, it works fine here.
[deep_mimic.zip](https://github.com/bulletphysics/bullet3/files/3905772/deep_mimic.zip)

Use this script to see that the joint positions are copied from the simulated humanoid to test humanoid, so obviously the joint angles cannot be 0.

~~~~~~~~~~~~~~~~~
python3 testrl.py --arg_file run_humanoid3d_backflip_args.txt
~~~~~~~~~~~~~~~~~

[image]

asawaswapnil, 2019-12-02:
Thanks a lot @erwincoumans  but the issue is still there. More concretely I mention my code snippet below. I don't know why the local angular velocities are coming to be 0

`		simJointStates = self._pybullet_client.getJointStatesMultiDof(self._sim_model, jointIndices)
		kinJointStates = self._pybullet_client.getJointStatesMultiDof(self._kin_model, jointIndices) 
		
		local_joints_orn=[self._pybullet_client.getEulerFromQuaternion(simJointStates[i][0]) if len(simJointStates[i][0])==4 else simJointStates[i][0]  for i in range(num_joints)]
		local_joints_ang_vel=[self._pybullet_client.getEulerFromQuaternion(simJointStates[i][1]) if len(simJointStates[i][1])==4 else simJointStates[i][1]  for i in range(num_joints)]
		print("local_joints_orn sim ",local_joints_orn,"\n")
		print("local_joints_ang_vel sim ",local_joints_ang_vel,"\n\n")

		local_joints_orn=[self._pybullet_client.getEulerFromQuaternion(kinJointStates[i][0]) if len(kinJointStates[i][0])==4 else kinJointStates[i][0]  for i in range(num_joints)]
		local_joints_ang_vel=[self._pybullet_client.getEulerFromQuaternion(kinJointStates[i][1]) if len(kinJointStates[i][1])==4 else kinJointStates[i][1]  for i in range(num_joints)]
		print("local_joints_orn kin ",local_joints_orn,"\n")
		print("local_joints_ang_vel kin ",local_joints_ang_vel,"\n\n")
`
here is my output at one of the iterations:

local_joints_orn sim  [(), (0.1152196397499862, 0.11387204772717362, 0.2977874526007448), (-0.08961941357230782, -0.010224382603043185, 0.08743547064494733), (0.0019748081766984503, 0.005854637072256874, -0.07421137981525736), (-1.514012414557107,), (0.2486054293717618, -0.47542793202109507, 0.07887917351288523), (-0.9373821318275208, -0.5878154320114033, -1.2173381197758961), (1.7255205560947862,), (), (0.1019996752182051, 0.013447284969755708, 0.27335096652733415), (-1.6907443211356996,), (0.49242872952718786, -0.559786733443045, 0.5851440493117894), (-0.08514776053902458, -0.07085052532022773, -0.0680510974039389), (1.2317630098509014,), ()] 

local_joints_ang_vel sim  [(), (0.7942265507176365, 1.2861881478527937, 1.6838846887534806), (-0.4948783633828217, -0.003736818688401964, 0.6422511641436564), (-0.2850388159558822, 0.11042926706234248, -2.102102227885315), (-1.232808578875529,), (-11.307387503844241, -4.622563805138032, -6.900018724378981), (-7.911489282616644, -0.15674205475840078, -5.215965094454368), (1.0782649319458029,), (), (0.4694809118170893, 0.23450057562490734, 0.17050649541504614), (-1.1972261894314569,), (2.153676376315211, -1.3610222266190486, -0.8457149986839578), (-0.7342520542017735, -0.2685803366152244, -0.06251173684758954), (1.163869588808136,), ()] 

local_joints_orn kin  [(), (0.09999643588782543, -0.07712711051587813, 0.008312900712608417), (0.0, -0.0, 0.0), (0.13650248434303422, -0.14280390877598167, 0.3338911423891141), (-0.37383426199272085,), (-0.08464039410593609, 0.04477646266402998, 0.1447528459938243), (-0.20166637158582204, 0.25895282850353973, -0.1329904782856438), (0.21228191805304036,), (), (0.16661706376628954, 0.03468464253173615, 0.24111634808927762), (-1.243688767639822,), (-0.02281216189379236, -0.20367265412272045, -0.2854190706241807), (0.22549000063826208, -0.17414243519266567, 0.03836070995503361), (0.4116865515585447,), ()] 

local_joints_ang_vel kin  [(), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0), (0.0,), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0), (0.0,), (), (0.0, 0.0, 0.0), (0.0,), (0.0, 0.0, 0.0), (0.0, 0.0, 0.0), (0.0,), ()] 

Note that here my model is not the pre-trained one so values of sim and kin are not near to identical.

asawaswapnil, 2019-12-02:
I think I now get it why was this happening. 

getJointStatesMultiDof gives output like:
((-0.010711879232444424, -0.003769487313739473, 0.0556463054096147, 0.9983859650971466), (-0.1750060589293822, 0.02596062223988028, 0.22054977257276773), (0.0, 0.0, 0.0, 0.0, 0.0, 0.0), (0.0, 0.0, 0.0)) 
0th index, i.e. jointPosition is of size 1 or 4. ( Pybullet guide mentions it to be of size 1 or 3!, please correct it)
1st index, i.e. jointVelocity if of size 1 or 3. ( I made a mistake here)( This is correct in Pybullet guide.)