bulletphysics/bullet3 #2507: Multiple simulators using EGL

guhur, 2019-11-26:
I am trying to run 10 different simulators in parallel. With `gpu=False`, the following code works fine, but with `gpu=True`, Bullet crashes.

Is there any reason why?

```
import pybullet as pb
from pybullet_utils.bullet_client import BulletClient
import pkgutil
from joblib import Parallel, delayed
from time import time

def load_pb_client(debug=False, gpu=True):
    client = BulletClient(pb.GUI if debug else pb.DIRECT)

    if gpu:
        egl = pkgutil.get_loader('eglRenderer')
        client.loadPlugin(egl.get_filename(), "_eglRendererPlugin")

    return client

def get_rgb(client_id):
    width, height, rgba, depth, mask = clients[client_id].getCameraImage(width=500, height=600)

clients = [load_pb_client(gpu=False) for i in range(10)]

start = time()
with Parallel(n_jobs=10, require='sharedmem') as parallel:
        parallel(delayed(get_rgb)(i) for i in range(5))
print(time() - start)
```

erwincoumans, 2019-11-26:
The EGL instances cannot be in the same thread. Try multiprocessing instead.
Here is an example, but I haven't tried using EGL with this yet:
https://github.com/erwincoumans/pybullet_robots/blob/master/batchsim3.py

closed 2019-11-26 by erwincoumans

guhur, 2019-12-05:
Ok thanks a lot ! :)