bulletphysics/bullet3 #2508: Replace LinearMath With GLM

kklouzal, 2019-11-27:
I propose replacing the `LinearMath` implementation with [`OpenGL Mathematics (GLM)`](https://github.com/g-truc/glm).

GLM is a very well established, feature rich, highly performant, header only, mathematics library.

A direct quote from their website:
> This library works perfectly with OpenGL but it also ensures interoperability with other third party libraries and SDK. It is a good candidate for software rendering (raytracing / rasterisation), image processing, **physic simulations** and any development context that requires a _simple and convenient mathematics library_.

Why?
- Reduce bullets overall codebase making it easier to manage, maintain, and improve upon in the future.
- Couple bullet more tightly with applications that are using it. Functions that accept arguments and return values will be directly interchangeable with applications that also use GLM for their mathematics.
- Performance. GLM boasts performance overall providing native internal SIMD support on mathematical operations.
- GLM is stable, future versions and updates promise to not break current implementations.
- C98 compatibility directly aligns with bullets goal of using a C-Library for it's low-level back-end.
- GLM has a _huge_ community of contributors dedicated to improving and maintaining the library which offers bullet a top-of-the-line mathematics library that is highly stable, feature rich, and bug free.

While I understand this would require a substantial amount of upfront work, the end result will provide many enhancements to the speed, usability, maintainability, and future improvement of bullet.

erwincoumans, 2019-11-27:
We haven't done much maintenance in LinearMath for the last 10 years. If we switch to a new library in the future, it will likely be Eigen or a differentiable math library.
Please re-open if someone has time to port all of Bullet to GLM, you could create a fork and show convincing benefits?

closed 2019-11-27 by erwincoumans

lyssdod, 2022-06-06:
I was recently looking into this and I have some questions related to porting. I was mainly interested in unifying SIMD optimizations and cleaning library's code a bit, because it has a lot of arch-specific SIMD ifdefs. Eigen definitely looks like a proper choice for optimizing LinearMath code; my main argument is backend support (a lot of archs), which should make bullet even faster. However:

- According to this: https://gitlab.com/libeigen/eigen/-/issues/2447, they're gonna switch to C++14 in the future (can be within few stable releases), and I've seen around the repo you've been trying to minimize this feature bump;
- They also appear to use STL for some things, which you've been against here (in the contribution guide);
- My main question is testing: imagine I hooked all relevant `LinearMath` code to Eigen, how can I run unittests to verify it's working correctly? Right now build fails with the following error:
```
$ mkdir build && cd build && cmake .. -DBUILD_BULLET2_DEMOS=OFF && make
...
Consolidate compiler generated dependencies of target Test_BulletInverseDynamicsJacobian
[ 83%] Built target Test_BulletInverseDynamicsJacobian
Consolidate compiler generated dependencies of target Test_BulletInverseForwardDynamics
[ 87%] Built target Test_BulletInverseForwardDynamics
Consolidate compiler generated dependencies of target Test_BulletInverseDynamics
[ 87%] Built target Test_BulletInverseDynamics
Consolidate compiler generated dependencies of target Test_PhysicsClientServer
[ 88%] Linking CXX executable Test_PhysicsClientServer
ld: library not found for -lBussIK
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [test/SharedMemory/Test_PhysicsClientServer] Error 1
make[1]: *** [test/SharedMemory/CMakeFiles/Test_PhysicsClientServer.dir/all] Error 2
make: *** [all] Error 2
```

Thank you!

lyssdod, 2022-09-02:
@erwincoumans what do you think?