bulletphysics/bullet3 #2510: deep mimic, position control and stable pd control

jhonsonlaid, 2019-11-29:
I'm testing deep mimic python code and find it initializes every joint with POSITION_CONTROL (in __init__()) and compute and applies forces with STABLE_PD_CONTROL.
And I tried to get joint states using ``getJointStatesMultiDof()``, it shows that joints' forces are zeros. 
However, if I comment the the POSITION_CONTROL codes in ``__init___()``, the joints' forces are meaningful numbers(use STABLE_PD_CONTROL only), but the humanoid model seems to freeze and dosen't discover action space.
 So why use POSITION_CONTROL first and use  STABLE_PD_CONTROL  after? And how can I  drive the model to discover the action space while getting meaningful forces?

erwincoumans, 2019-11-29:
How did you install pybullet? Did you use pip/pip3 or python3 setup.py build/install?
Does this work for you?
~~~~~~~~~~~~~~~~~~~~~~~
python3 -m pybullet_envs.deep_mimic.testrl --arg_file run_humanoid3d_walk_args.txt
~~~~~~~~~~~~~~~~~~~~~~~

closed 2019-11-29 by erwincoumans

jhonsonlaid, 2019-12-02:
Thank you for your reply.
1. the forces is applied by https://github.com/bulletphysics/bullet3/blob/d2665d3d80b89ef36fad49a8479bd0dd85f65326/examples/pybullet/gym/pybullet_envs/deep_mimic/env/humanoid_stable_pd.py#L350 ``setJointMotorControlMultiDofArray``  with ``controlMode=STABLE_PD_CONTROL`` , which I have no idea where to get the force number.

2. ``python3 -m pybullet_envs.deep_mimic.testrl --arg_file run_humanoid3d_walk_args.txt`` work for me.

erwincoumans, 2019-12-02:
Those forces are computer in the few lines right above the call to setJointMotorControlMultiDofArray .

jhonsonlaid, 2019-12-02:
Aren't  the forces passed to setJointMotorControlMultiDofArray the MaxForces (used to clip) instead of the exact force value.

erwincoumans, 2019-12-02:
Ah yes, you need to enable usePythonStablePD (in pybullet_deep_mimic_env.py) to inspect the forces from Python. We need to expose them explicitly when computed in C++/STABLE_PD_CONTROL.

jhonsonlaid, 2019-12-03:
Thanks for explanation. All problems solved. :-)