bulletphysics/bullet3 #2511: [PyBullet] Fixed joint using constraint is moving

nordmoen, 2019-11-29:
Hi, I'm trying to simulated modular robots with a variable amount of modules. 

I have set everything up and it is working quite nice, however, I have problem replicating results from one machine to another (e.g. the same robot and controller, pickled with Python, does not achieve the same performance with different compilers). Trying to track this down I realized that bodies connected with `createConstraint` and `JOINT_FIXED` are allowed to move quite a lot which is probably exasperating the problem. 

Since the modules are fixed, they come as URDFs, and there needs to be a variable amount I can't see a different solution than to connect them using `createConstraint`. I found this link: https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=12758 which suggest to instead create compound URDFs, however, that seems strange as the constraint should be fixed. Is there anything I can do to get a fixed joint between two different URDF models?

erwincoumans, 2019-11-29:
If there are large differences in effective mass, constraint maybe not strong enough. We probably need to use a different solving method to fix this. 

Instead of createConstraint, you can combine two (or more) urdf files, using the Python [urdf editor](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_utils/urdfEditor.py).
See an example here, connecting a Kuka arm to a Husky robot:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_utils/examples/combineUrdf.py

~~~~~~~~~~~~~~~~~~
python3 -m pybullet_utils.examples.combineUrdf
~~~~~~~~~~~~~~~~~~

This editor loads two urdf files (in a temporary dynamics world) and create a combined urdf programmatically using createMultiBody.

closed 2019-11-29 by erwincoumans

cmuell89, 2021-02-03:
@erwincoumans Is the URDF join method easily reversible? Other sims stacks (Gazebo/MoveIt) expose an "attach object" method so that we can fake an effective pick/grasp with an end-effector. Ideally, there is also a detachObject method. The PyBullet 'createConstraint' methods works in some cases, but in a situation where we want to measure force feedback on the object without it violating the constraint or being released from the end-effector, a URDF fusing approach might be more ideal. It there an easy way to "detach" or "unjoin" the two urdfs?