bulletphysics/bullet3 #2513: Question about Forward Dynamics in stepSimulation()

gsutanto, 2019-12-02:
Dear Erwin and PyBullet authors,

If the inverse dynamics via Recursive Newton-Euler Algorithm (RNEA) is in summary performing:
torque = H(q) qdd + C(q, qd) qd + G(q)
where q is joint position, qd is joint velocity, qdd is joint acceleration, H(q) is inertia matrix, C(q, qd) is Coriolis matrix, and G(g) is gravity vector.

What are the forward dynamics computation internally done by PyBullet after invoking:
sim.setJointMotorControlArray()
sim.stepSimulation()
(setRealTimeSimulation is turned off)
for a manipulator doing a free-space motion without contacts?

Is it:
next_qdd = (H(q))^{-1} (torque - C(q, qd) qd - G(q) - damping_constant*qd)
or are there any additional term that I am missing in the above forward dynamics computation I wrote?

Also what is the integration scheme that PyBullet uses? Is it Euler integration or Runge-Kutta method, or other integration method?

Best Regards,

Giovanni Sutanto

erwincoumans, 2019-12-03:
The forward dynamics and inverse dynamic agree with each other indeed.
You can check [invdyn_bullet_comparison](https://github.com/bulletphysics/bullet3/blob/master/Extras/InverseDynamics/invdyn_bullet_comparison.cpp) for details on this, there is a unit test to make sure of this.
In stepSimulation, we compute the unconstrained accelerations (next_qdd), then we compute the constraint forces (if any), and then perform an integration step. By default, we use a Symplectic Euler integrator (v_t+1 = v_t+dt*a_t), and then x_t+1 += dt * v_t+1.

gsutanto, 2019-12-04:
Dear @erwincoumans ,

I wrote the following code, trying to match forward dynamics computation of PyBullet:
```
import pybullet_utils.bullet_client as bc
import pybullet
import pybullet_data

import os, time, copy
import numpy as np

np.set_printoptions(precision=5, suppress=True)

hz = 240.0
dt = 1.0 / hz

sim = bc.BulletClient(connection_mode=pybullet.GUI)

sim.setGravity(0, 0, -9.81)
sim.setTimeStep(dt)
sim.setRealTimeSimulation(0)

controlled_joints = [0, 1, 2, 3, 4, 5, 6]
addtnl_torque = np.array([0,0,0,1.0,0,0,0])
n_dofs = len(controlled_joints)

pybullet.setAdditionalSearchPath(pybullet_data.getDataPath())
urdf_path = "kuka_iiwa/model.urdf"

world_id = pybullet.loadURDF("plane.urdf", [0, 0, 0])
robot_id = sim.loadURDF(urdf_path, basePosition=[0.0, 0.0, 0.0], 
                        useFixedBase=True,
                        flags=pybullet.URDF_USE_INERTIA_FROM_FILE)

full_urdf_path = pybullet_data.getDataPath()+'/'+urdf_path
print("full_urdf_path = ", full_urdf_path)

def getCurrentJointPosVel():
    cur_joint_states = sim.getJointStates(robot_id, controlled_joints)
    cur_joint_pos = [cur_joint_states[i][0] for i in range(n_dofs)]
    cur_joint_vel = [cur_joint_states[i][1] for i in range(n_dofs)]
    return cur_joint_pos, cur_joint_vel

def getCurrentEEVel():
    ee_state = sim.getLinkState(robot_id, n_dofs-1,
    	                        computeLinkVelocity=True, 
                                computeForwardKinematics=True)
    ee_lin_vel = np.array(ee_state[6]) # worldLinkLinearVelocity
    ee_ang_vel = np.array(ee_state[7]) # worldLinkAngularVelocity
    return ee_lin_vel, ee_ang_vel

def computeGravityCompensationControlPolicy():
    [cur_joint_pos, _] = getCurrentJointPosVel()
    grav_comp_torque = sim.calculateInverseDynamics(robot_id, cur_joint_pos, 
                                                    [0] * n_dofs, 
                                                    [0] * n_dofs)
    return np.array(grav_comp_torque)

def computeRBDInertiaMatrixAndNonLinearEffects(joint_pos, joint_vel):
    rbd_inertia_matrix = sim.calculateMassMatrix(robot_id, list(joint_pos))
    rbd_non_linear_effects = sim.calculateInverseDynamics(robot_id, joint_pos, joint_vel, [0.0] * 7)
    return np.asarray(rbd_inertia_matrix), np.asarray(rbd_non_linear_effects)

def computeForwardDynamics(joint_pos, joint_vel, torque, is_using_damping):
     [rbd_inertia_matrix, rbd_non_linear_effects
      ] = computeRBDInertiaMatrixAndNonLinearEffects(joint_pos, joint_vel)

     temp = torque - rbd_non_linear_effects
     
     if is_using_damping:
[... 59 lines cut ...]
q   =  [ 0.01434  0.43819  0.00456  1.56898  0.0394   0.80786 -0.20734]
qd  =  [   0.20045    0.23988    0.19792    1.55019    6.84358    1.43943    -100.     ]
qdd =  [    98.30342     -3.26628     94.7481      54.91427   3286.69069   142.19211 -48000.     ]

computed_qdd_wo_damping =  [-0.12334873929538803, -0.31044579201076894, 1.34318877004002, 1.844418856173523, -1.859091087725854, 71.87495287412763, 14.250051847263997]
computed_qdd_w_damping =  [-98.60609334917842, 2.330701867992183, -94.60720492133593, -54.46046220217467, -3261.883903184632, -142.6948281249777, 52428.17894564496]
time_count =  298
q   =  [0.01347 0.43923 0.00374 1.57449 0.01123 0.81137 0.20932]
qd  =  [ -0.21043   0.24955  -0.19648   1.32265  -6.75987   0.84374 100.     ]
qdd =  [  -98.61101     2.31983   -94.65697   -54.61012 -3264.82891  -142.9671   48000.     ]

computed_qdd_wo_damping =  [-0.5576537769445713, -0.37985028752547617, 0.7348986795035409, 1.9251134484523313, 3.306386283072615, 71.88455616903151, -11.745442115008972]
computed_qdd_w_damping =  [98.84740646842464, -3.156979453929905, 94.50642393197342, 54.19514682238956, 3240.144601754793, 139.33355994463278, -52401.87002323955]
time_count =  299
q   =  [ 0.01431  0.44021  0.00456  1.58094  0.03937  0.81731 -0.20734]
qd  =  [   0.20145    0.23633    0.1975     1.54828    6.75262    1.42493  -100.     ]
qdd =  [    98.85238     -3.17111     94.55403     54.15297   3242.99806  139.4855    -48000.     ]
```
As you see, the qdd does NOT match perfectly with computed_qdd_w_damping. What could have been missing here?

P.S.: Can you please re-open this issue? I cannot re-open it from my side. Thanks.

Best Regards,

Giovanni

erwincoumans, 2019-12-04:
I pointed you to the c++ comparison in my previous post that shows that inverse dynamics and forward dynamics perfectly matches. Please check that first.

gsutanto, 2019-12-04:
Dear @erwincoumans ,

Yes, I checked the C++ comparison that you pointed out. Thanks for the pointer, if I added the following line to turn off all dampings:
```
for link_idx in range(8):
    sim.changeDynamics(robot_id, link_idx, linearDamping=0.0, angularDamping=0.0, jointDamping=0.0)
```
I get:
```
computed_qdd_wo_damping =  [-0.09149939522323809, -1.1269857321533738, -0.04343543439520839, 0.2749570124290451, 1.5273812974757208, -4.299123801721428, 3.68235490199738]
time_count =  185
q   =  [ 0.05478  0.48704 -0.0144   1.61118  0.26899  1.50081 -0.06796]
qd  =  [ 0.0282   0.67607 -0.02073  2.85999  0.8556   1.71583  0.50151]
qdd =  [-0.0915  -1.12699 -0.04344  0.27496  1.52738 -4.29912  3.68235]

computed_qdd_wo_damping =  [-0.08963881440774904, -1.1422114297381194, -0.0449198649028031, 0.2841542759810468, 1.5229237681442909, -4.325861277283076, 3.7113760734549426]
time_count =  186
q   =  [ 0.0549   0.48983 -0.01449  1.6231   0.27259  1.50789 -0.06581]
qd  =  [ 0.02783  0.67131 -0.02092  2.86118  0.86194  1.69781  0.51697]
qdd =  [-0.08964 -1.14221 -0.04492  0.28415  1.52292 -4.32586  3.71138]
```
So qdd matches with computed_qdd_wo_damping

But if I only turn off link damping (but keep the joint damping) with the following line:
```
for link_idx in range(8):
    sim.changeDynamics(robot_id, link_idx, linearDamping=0.0, angularDamping=0.0)
```
I get:
```
computed_qdd_w_damping =  [85.07159426563172, -5.058375451034304, 95.43037731605654, 67.34080965632552, 4067.8006510317437, 190.98872769207105, -53430.94839256632]
time_count =  267
q   =  [ 0.02156  0.40642  0.01124  1.40895  0.04783  0.64232 -0.20057]
qd  =  [   0.17547    0.28522    0.20324    1.61508    8.4716     1.63709
 -100.     ]
qdd =  [    85.07159     -5.05838     95.43038     67.34081   4067.80065
    190.98873 -48000.     ]

computed_qdd_w_damping =  [-85.62848882336172, 4.139339274962428, -95.458061096132, -67.04714357909292, -4043.5329943306865, -189.97524176156676, 53399.68641337165]
time_count =  268
q   =  [0.0208  0.40768 0.01043 1.41451 0.01293 0.64585 0.2161 ]
qd  =  [ -0.18132   0.30247  -0.1945    1.33571  -8.37646   0.84553 100.     ]
qdd =  [  -85.62849     4.13934   -95.45806   -67.04714 -4043.53299  -189.97524
 48000.     ]
```
As you can see, most of qdd matches with computed_qdd_w_damping, except the last joint, where it stuck at 48000 (some magical number?). Is 48000 some kind of maximum acceleration possible?

Thanks so much for your quick response!

Best Regards,

Giovanni

erwincoumans, 2019-12-04:
By default Bullet btMultiBody clamps the velocity too the range [-100,100]. Usually if the velocity exceeds this, the simulation would go out-of-control and feedback loops go into NaN.

You can change this threshold using
~~~~~~~~~~~
pybullet.changeDynamics(bodyUid, -1, maxJointVelocity=newvalue) 
~~~~~~~~~~~
to become [-newvalue, newvalue]

gsutanto, 2019-12-05:
Dear @erwincoumans ,

If I do:
```
for link_idx in range(8):
    sim.changeDynamics(robot_id, link_idx, linearDamping=0.0, angularDamping=0.0)
    sim.changeDynamics(robot_id, link_idx, maxJointVelocity=200)
```
The printout is:
```
computed_qdd_w_damping =  [239.40159838813173, -12.851929098061982, 127.11323280006287, -21.25870796776021, 100.73439036935342, 22.316377516515047, -100286.46707253941]
time_count =  215
q   =  [ 0.07018  0.326    0.06283  1.09724  0.08877  1.5479  -0.35772]
qd  =  [   0.51409    0.33366    0.28739    1.36155    0.22052    0.33028
 -200.     ]
qdd =  [   239.4016     -12.85193    127.11323    -21.25871    100.73439
     22.31638 -96000.     ]
```
The qd of last joint is clamped at -200, but now the qdd of last joint is also clamped at -96000 (twice the value it used to be before). It seems that the qdd limit is a constant multiply of the qd limit?

We can close this issue for now.

Thanks!

Best Regards,

Giovanni Sutanto

closed 2019-12-05 by gsutanto

deepakraina99, 2020-07-14:
Have you found a solution to this issue? I am having the same issue.