bulletphysics/bullet3 #2515: When the gripper touch the object, the object going down to the table

Alchemist77, 2019-12-03:
Hi,

Thank you for supporting pybullet.

I already made a environment that stable object movement with this code
p.changeDynamics(boxId,-1,lateralFriction=20, contactStiffness = 20, contactDamping=20)
However, I had a problem when the gripper touch for grasping the object, the object goes down to the table. of course not all area... some part of the area, for example
original box_z = 0.87
box_z after gripper touches = 0.84

Do you know the function that the object never goes into the table? 

Thank you.

erwincoumans, 2019-12-09:
You should not make contact too soft, and also make sure the robot/gripper doesn't push too strongly against the object. Otherwise you need to tweak the constraint solver iterations and smaller time step and then performance becomes a problem.

closed 2019-12-09 by erwincoumans

ZainBashir, 2019-12-10:
Thank you for your reply

Is there way to make contact like soft in urdf??? I made a simple block like this

```
<robot name="block">
  <link name="block">

    <inertial>
      <origin xyz="0 0 0" />
      <mass value="0.2" />
      <inertia  ixx="1" ixy="0.0"  ixz="0.0"  iyy="1"  iyz="0.0"  izz="1" />
    </inertial>

    <visual>
      <origin xyz="0 0 0"/>
      <geometry>
        <box size="0.085 0.085 0.085" />
      </geometry>
      <material name="red">
        <color rgba="0.4 0.4 1.0 1.0"/>
      </material>
    </visual>

    <collision>
      <origin xyz="0 0 0"/>
      <geometry>
        <box size="0.086 0.086 0.086" />
      </geometry>
    </collision>

  </link>
</robot>
```

Is it enough that robot can grasp?

Alchemist77, 2019-12-10:
Thank you for your reply

Is there way to make contact like soft in urdf??? I made a simple block like this

```
<robot name="block">
  <link name="block">

    <inertial>
      <origin xyz="0 0 0" />
      <mass value="0.2" />
      <inertia  ixx="1" ixy="0.0"  ixz="0.0"  iyy="1"  iyz="0.0"  izz="1" />
    </inertial>

    <visual>
      <origin xyz="0 0 0"/>
      <geometry>
        <box size="0.085 0.085 0.085" />
      </geometry>
      <material name="red">
        <color rgba="0.4 0.4 1.0 1.0"/>
      </material>
    </visual>

    <collision>
      <origin xyz="0 0 0"/>
      <geometry>
        <box size="0.086 0.086 0.086" />
      </geometry>
    </collision>

  </link>
</robot>
```

Is it enough that robot can grasp? 
Also, I see a lot of slippery motion when the gripper grasp object(I added later friction and very slow step simulation) Could you suggestion some function to protect it?

Thank you