bulletphysics/bullet3 #2525: proper way to get friction force in btMultibody

Rullec, 2019-12-06:
Hi All,
  Thanks for your time in this issue. 
  I am truly a beginner in bullet and have less knowledge for it.Currently I am using **btMultibody** to construct a robot with a floating base. At this moment I can get the the global pos & force applied in these contact points from class btPersistentManifold and btCollisionObject. But now I still **wonder the friction force in these contact points**. I have finished both of the Quick start and the User Manual and the classic RigidBodyDynamic book by featherstone(2008), also some tutorials for generalized coordinates and Lagragian Dynamics, but It seems no answer can I found.
  How can I get these friction force applied between collision objects from bullet? I got trapped in this questions for couple of days. 
  Thanks all tips and help in any degree!

Steven89Liu, 2019-12-09:
the friction related constraints are processed by these functions, addMultiBodyTorsionalFrictionConstraint() and addMultiBodyFrictionConstraint().  Could you attach your simple test? then we can work together to figure out what happened there. thanks.

erwincoumans, 2019-12-09:
PyBullet reports the contact normal force and also lateral friction forces/directions, you can use it as sample code:

https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/PhysicsServerCommandProcessor.cpp#L7479

~~~~~~~~~~~~~~~~
for (int p = 0; p < manifold->getNumContacts(); p++)
					{
						b3ContactPointData pt;
						pt.m_bodyUniqueIdA = objectIndexA;
						pt.m_bodyUniqueIdB = objectIndexB;
						const btManifoldPoint& srcPt = manifold->getContactPoint(p);
						pt.m_contactDistance = srcPt.getDistance();
						pt.m_contactFlags = 0;
						pt.m_linkIndexA = linkIndexA;
						pt.m_linkIndexB = linkIndexB;
						for (int j = 0; j < 3; j++)
						{
							pt.m_contactNormalOnBInWS[j] = srcPt.m_normalWorldOnB[j];
							pt.m_positionOnAInWS[j] = srcPt.getPositionWorldOnA()[j];
							pt.m_positionOnBInWS[j] = srcPt.getPositionWorldOnB()[j];
						}
						pt.m_normalForce = srcPt.getAppliedImpulse() / m_data->m_physicsDeltaTime;
						pt.m_linearFrictionForce1 = srcPt.m_appliedImpulseLateral1 / m_data->m_physicsDeltaTime;
						pt.m_linearFrictionForce2 = srcPt.m_appliedImpulseLateral2 / m_data->m_physicsDeltaTime;
						for (int j = 0; j < 3; j++)
						{
							pt.m_linearFrictionDirection1[j] = srcPt.m_lateralFrictionDir1[j];
							pt.m_linearFrictionDirection2[j] = srcPt.m_lateralFrictionDir2[j];
						}
						m_data->m_cachedContactPoints.push_back(pt);
					}
~~~~~~~~~~~~~~~~

closed 2019-12-09 by erwincoumans

Rullec, 2019-12-10:
thanks all of you a lot! Sorry for my late reponse, I will do these tests at once:)

erwincoumans, 2019-12-10:
The friction and normal forces of contacts are stored in the contact points. Check my previous reply.