bulletphysics/bullet3 #2531: The collision shape problem of concave object

TFLQW, 2019-12-09:
I have got the urdf model of a hole from solidworks, and also a mesh file of the hole. But when i try to assemble the peg into the hole. I found the collision shape of the hole is cylinder which is solid not contain the hole of the cylinder.
Could you tell me how to confige this stl model or use which of software to config this property

TFLQW, 2019-12-09:
this is that model stl
[hole_f.zip](https://github.com/bulletphysics/bullet3/files/3937525/hole_f.zip)

erwincoumans, 2019-12-09:
You need to create a convex decomposition or crease a SDF file of the mesh. We need to document this better. can you share the URDF file too?

closed 2019-12-09 by erwincoumans

RafaelsNeurons, 2024-01-23:
Hey :)
@erwincoumans, could you tell me where I can find the GenerateSDF.exe ?