bulletphysics/bullet3 #2532: Column limit for code style (clang format)

Lawrencemm, 2019-12-09:
Hi, there are some parts of the code that I'm finding really difficult to read because of excessive line length.

I'm proposing setting ColumnLimit in the clang format file to something sane like maybe 120 though personally I'd prefer 100 or even 80.

erwincoumans, 2019-12-09:
There is a clang-format file, so you can adjust it and run clang-format so it suits your needs.

closed 2019-12-09 by erwincoumans

Lawrencemm, 2019-12-09:
Yeah I've started doing that in CLion which is helping, just hit control alt L and it reformats with my modified format file.

I don't have such good eyesight so I can't really fit long lines on any screen without getting rsi in my neck :face_with_head_bandage:

mreff555, 2022-12-29:
Actually, the google style guide recommends a maximum of 80 characters. 
[Google Style Guide - Line length](https://google.github.io/styleguide/cppguide.html#Line_Length)