bulletphysics/bullet3 #2555: Quaternion angle calculation loss of precision

RubendeBruin, 2019-12-24:
In btQuaternion.h the function "btScalar getAngle() const" calculates the angle from the inverse cosine of w.
This gives numerical issues when the angle is almost zero (~1e-8 in case of double). In that case the w term becomes almost 1 in which case a lot of significant digits are lost ( w = 0.9999999999999998) and the calculation of the angle becomes less precise than possible.

This can be solved by calculating the angle from the atan2 instead:

```
btVector3 xyz{ q[0], q[1], q[2] };
double w{ q[3] };

double n = xyz.norm();
double angle{ 2 * atan2(n, abs(q[3])) };
if (w < 0)
{
    angle = -angle;
}
```
This approach can be found in the Eigen and scipy (python) packages.

erwincoumans, 2020-01-09:
Thanks for the report, if there is time (and motivation) we may improve this eventually (not urgent, unless you have a reproduction example that shows issues).

GregSlazinski, 2020-05-14:
I'd like to concur. I've dealt with this problem and loss of precision is tremendous, especially if you're using 32-bit precision floats.

Example if you're trying to convert angle to cos, and back to angle from cos, you lose a lot.
   float a1=0.0002f;
   float c=cosf(a1); // gives 1.0f
   float a2=acosf(c); // gives 0.0f

See here how I'm handling this in my game engine:
https://github.com/Esenthel/EsenthelEngine/blob/master/Engine/Source/Math/Quaternion.cpp#L61

closed 2021-04-26 by erwincoumans