bulletphysics/bullet3 #2559: Training in a minitaur environment

shaosb, 2019-12-30:
I'm trying to reproduce the content of this article in Sim-to-Real Learning Agile Locomotion For Quadruped Robots, and I expect that I will get a gait similar to the video in YouTube. I did not find how to train such a gait in the open source code, only the parameters after training. So is there any code for training these parameters that can be used, if not, what tricks can be used in the process of building a PPO network and training PPO? I tried training on a PPO algorithm many times and did not reach results similar to those in YouTube videos. I used the two training environments of minitaur_reactivev_env.py and miniatur_trotting_env.py in minitaur.

erwincoumans, 2019-12-31:
1) You can also use PPO2 from stable baselines, see their pre-trained policy here:
https://github.com/araffin/rl-baselines-zoo

2) You can also use estool, based on CMA-ES, as described here:
https://github.com/hardmaru/estool
it also has pre-trained weights.

1) We used TF Agents PPO for the original paper, which is included with PyBullet. The policy sometimes differs dependent on the random initialization weights.

Please describe what kind of motion you get if above doesn't help.

closed 2019-12-31 by erwincoumans

IverYangg, 2022-04-19:
Hi, i want to do the same as yours? have you ever train a minitaur in pybullet with PPO?