bulletphysics/bullet3 #2562: Render should return a uint8 dtype

mimoralea, 2019-12-31:
To make PyBullet environments compatible with the OpenAI Gym Video Recorder (and Monitor) wrappers, the PyBullet base environment `render` function should return a uint8 dtype. 

Here is the relevant line in the PyBullet base environment:
https://github.com/bulletphysics/bullet3/blob/c9a0646bdae6c4f263b3affc9908644eaca46cff/examples/pybullet/gym/pybullet_envs/env_bases.py#L97

We likely need to change `rgb_array = np.array(px)` to `rgb_array = np.array(px, dtype=np.uint8)`. Or perhaps, `return rgb_array` to `return rgb_array.astype(np.uint8)`.

Otherwise, we get an error similar to this one: https://stackoverflow.com/questions/57740328/warn-tried-to-pass-invalid-video-frame-marking-as-broken-your-frame-has-data.

Here is the line in the Video Recorder wrapper that checks for the dtype of the frame generated by the `render` function call: https://github.com/openai/gym/blob/c33cfd8b2cc8cac6c346bc2182cd568ef33b8821/gym/wrappers/monitoring/video_recorder.py#L299

erwincoumans, 2020-01-02:
Thanks for the report, is should be fixed in this pull request:
https://github.com/bulletphysics/bullet3/pull/2564

mimoralea, 2020-01-02:
Super, thanks!

closed 2020-01-02 by mimoralea

erwincoumans, 2020-01-02:
Note that the default CPU renderer is very slow compared to OpenGL.
One option is to run the physics server with GUI before enjoying the Gym environments:
~~~~~~~~~~~
python3 -m pybullet_envs.examples.runServer&
~~~~~~~~~~~
By default, pybullet envs try to connect to already running servers, which can render using hardware OpenGL.
For PyBullet 2.6.2 another option is to export environment variable PYBULLET_EGL=1, then it will try to use EGL.

Running the server (runServer) is preferred.