bulletphysics/bullet3 #2571: changeVisualShape does not work with eglRenderer

liruiw, 2020-01-07:
Hello,

I'm having an issue with changeVisualShape with loaded texture in Direct Mode. The images returned by getCameraImage seems not changing, while it's updated in GUI mode and TinyRenderer. Any ideas why it could be happening? 

Also, is there APIs to change ambient light intensities in Direct Mode? 

Thanks!

bibbygoodwin, 2020-05-25:
Hi @liruiw, did you resolve this at any point? I'm not using loaded textures, just standard geoms from an .mjcf file, but believed that calling changeVisualShape would work. For me, this method appears to have no effect no matter which renderer I use (ER_BULLET_HARDWARE_OPENGL or ER_TINY_RENDERER)

closed 2020-05-27 by liruiw

ycheng517, 2020-06-01:
This is still an issue, see possibly related ticket: https://github.com/bulletphysics/bullet3/issues/2777