bulletphysics/bullet3 #2578: Player presence breaks physics

TaiTair, 2020-01-10:
Please refer to this page: http://mcro.org/issues/view_issue/2433

Physics should not be affected by the presence of a player.

erwincoumans, 2020-01-12:
This is too bad and unlikely due to Bullet. If they can create a small reproduction in Bullet or PyBullet we can look at it.

closed 2020-01-12 by erwincoumans

TaiTair, 2020-01-14:
Well the developers of the Game are blaming it on the physics engine they are using: Bullet. 

On that link I shared you can see the comments by "Beginner" where he clearly blames Bullet.

Could you possibly reach out as one of the Bullet Developers and tell them to get their house in order?

deltars, 2020-01-31:
Hi @erwincoumans 

I am the lead developer of Stormworks, we met at GDC last year.

My apologies that one of our players has brought the issue here. As you rightly point out, a simple reproduction case would be required to be able to look at this, and thank you for offering to help with a simple reproduction.

We are sure that in some cases, the physics simulation of some constraints resolves differently in a particular circumstance. This may be triggered by the presence of a ghost object, which we attach to the player.

However, this issue only occurs with certain constraint parameters and masses. It will be of no surprise to you or any one ever working with real time physics, that some settings and configurations give more stable results than others. Therefore, I would say this is not a defect. We do allow players to tweak some of these values themselves, and in this case, some players have found a setting that works less well.

I want to commend Erwin and others involved in the creation of bullet physics. The library is truly superb and holds up incredibly well to our players throwing the most challenging and unusual problems. It has also proven to be incredibly fast at resolving complex collisions and simulations. Thank you.

TaiTair I appreciate your passion for Stormworks, but please consider your expectations. Bullet is incredibly well engineered, and Stormworks has been allowed to use it for free. It simply isn't right for you to tell Bullet to "get their house in order" when your game doesn't behave as you personally expect.

Discussed in full at:

http://mcro.org/issues/view_issue/2433

Regards

TaiTair, 2020-02-07:
@deltars 

The fact you are telling me I am not right in telling Bullet to "get their house in order" proves how much the point flew above your head.
I didn't tell them to do that. I told them to confirm that the problem is on your end and that YOU (deltars and stormworks devs alike) should get your house in order.

It's not that it doesn't behave as I personally expect. It's that it doesn't behave correctly. There is a bug you are refusing to repair. Go ahead and bury your head in the sand, I don't care. Already gave your game a negative review anyway. I had passion for it but you did a good job of dispelling that.