bulletphysics/bullet3 #2579: The btGjkPairDetector fails to detect collision between primitive sphere and primitive box.

Levi-Armstrong, 2020-01-11:
I have been using Bullet as a collision checker for robotic motion planning and I have noticed that sometime it fails to detect a collision between objects. Below is one case that can be used to confirm it fails. Is this a result of the size of the objects?

Sphere:
Radius: 0.01
Position:  0.0197195 , -1.03115 , 0.37504

Box:
Side length: 0.005
Position: 0.0275 , -1.0275 , 0.3725

Levi-Armstrong, 2020-01-11:
I rebuilt bullet with double precision turn on and now everything works. Are there known issues with using double precision?

erwincoumans, 2020-01-11:
Double precision is preferred. For our robotics and machine learning research we exclusively use double precision version of PyBullet (using Bullet under the hood), since it is more precise.

Also, What collision margins do you use? The relative sizes (-1.03115 and 0.005) could be a challenge for single precision.

erwincoumans, 2020-01-12:
If you have a small reproduction case for double, let us know and re-open. Thanks!

closed 2020-01-12 by erwincoumans

Levi-Armstrong, 2020-01-12:
Thank you for the feedback, I will create test only using Bullet components and post here.