bulletphysics/bullet3 #2586: getKeyboardEvents does not work with shared memory server

ebronstein-symbio, 2020-01-16:
Getting keyboard events using `getKeyboardEvents` works for me in `GUI` mode, but it does not work when I launch the shared memory physics server using the `App_PhysicsServer_SharedMemory_GUI_gmake_x64_release` executable. Calling `getKeyboardEvents` just returns an empty dictionary.

Here is a minimal example:

1. Launch `App_PhysicsServer_SharedMemory_GUI_gmake_x64_release`. The GUI appears.
2. Execute the following code.
```
import time
import pybullet as pb

pb.connect(pb.SHARED_MEMORY)

while True:
    print(pb.getKeyboardEvents())
    time.sleep(0.5)
```

The code repeatedly prints `{}`.

erwincoumans, 2020-01-16:
~~~~~~~~~~~~~
python3 -m pybullet_envs.examples.runServer&
~~~~~~~~~~~~~
Then run your script, it reports key presses (as long as that GUI window is in focus when you press keys).

closed 2020-01-16 by erwincoumans

ebronstein-symbio, 2020-01-16:
Thank you! That worked for me.