bulletphysics/bullet3 #2588: pybullet documentation - clarification regarding torque control

GrauleM, 2020-01-16:
First, thanks for the great work you have done and are doing on bullet and pybullet!

According to the PyBullet Quick Start Guide, the implementation of `TORQUE_CONTROL` in  `setJointMotorControl2` relies on external forces/torques. External forces only work when explicitly stepping the simulation.

I assume that `TORQUE_CONTROL` therefore also only works when explicitly stepping the simulation. If this is correct, I think it would be helpful to add a quick note stating this somewhere near the section on `setJointMotorControl2`, such that this info is easier to find.

erwincoumans, 2020-01-16:
Thanks! Yes, this is correct. We should update the doc.

erwincoumans, 2020-01-24:
Added a comment in the section:
TORQUE_CONTROL will apply a torque instantly, so it only is effective when calling stepSimulation explicitly.

closed 2020-01-24 by erwincoumans