bulletphysics/bullet3 #2590: Anisometric planar friction

LemonPi, 2020-01-20:
Friction between surfaces seem to depend on their relative orientation. My experiment is to apply the same external force (in link frame) to a block and look at the change in position and yaw of the block in link frame as it yaws with respect to the floor.
[image]
I'm not sure how friction works in bullet - is there some issue with alignment of particles? The periodicity in the dynamics where we get the least dyaw every 45 degrees makes me suspect something like this is happening. If so, how can I resolve this?

Reproducible code (depends on numpy)
```python
import logging
import math
import pybullet as p
import time

import numpy as np
import pybullet_data
from matplotlib import pyplot as plt

logger = logging.getLogger(__name__)
logging.basicConfig(level=logging.DEBUG,
                    format='[%(levelname)s %(asctime)s %(pathname)s:%(lineno)d] %(message)s',
                    datefmt='%m-%d %H:%M:%S')
logging.getLogger('matplotlib.font_manager').disabled = True

def rotate_wrt_origin(xy, theta):
    return (xy[0] * math.cos(theta) - xy[1] * math.sin(theta),
            xy[0] * math.sin(theta) + xy[1] * math.cos(theta))

def angular_diff(a, b):
    """Angle difference from b to a (a - b)"""
    d = a - b
    if d > math.pi:
        d -= 2 * math.pi
    elif d < -math.pi:
        d += 2 * math.pi
    return d

def get_dx(px, cx):
    dpos = cx[:2] - px[:2]
    dyaw = angular_diff(cx[2], px[2])
    dx = np.r_[dpos, dyaw]
    return dx

def dx_to_dz(px, dx):
    dz = np.zeros_like(dx)
    # dyaw is the same
    dz[2] = dx[2]
    dz[:2] = rotate_wrt_origin(dx[:2], px[2])
    return dz

init_block_pos = [0.0, 0.0]
init_block_yaw = -0.

physics_client = p.connect(p.GUI)
p.setTimeStep(1. / 240.)
p.setRealTimeSimulation(False)
p.setAdditionalSearchPath(pybullet_data.getDataPath())

blockId = p.loadURDF("./block_big.urdf", tuple(init_block_pos) + (-0.02,),
                     p.getQuaternionFromEuler([0, 0, init_block_yaw]))
planeId = p.loadURDF("plane.urdf", [0, 0, -0.05], useFixedBase=True)
p.resetDebugVisualizerCamera(cameraDistance=0.5, cameraYaw=0, cameraPitch=-85,
                             cameraTargetPosition=[0, 0, 1])

STATIC_VELOCITY_THRESHOLD = 1e-6

def _observe_block(blockId):
    blockPose = p.getBasePositionAndOrientation(blockId)
    xb = blockPose[0][0]
    yb = blockPose[0][1]
    roll, pitch, yaw = p.getEulerFromQuaternion(blockPose[1])
    return np.array((xb, yb, yaw))

def _static_environment():
    v, va = p.getBaseVelocity(blockId)
    if (np.linalg.norm(v) > STATIC_VELOCITY_THRESHOLD) or (
            np.linalg.norm(va) > STATIC_VELOCITY_THRESHOLD):
        return False
    return True

p.setGravity(0, 0, -10)
# p.changeDynamics(blockId, -1, lateralFriction=0.1)
p.changeDynamics(planeId, -1, lateralFriction=0.5, spinningFriction=0.3, rollingFriction=0.1)
F = 200
MAX_ALONG = 0.075 + 0.2

for _ in range(100):
    p.stepSimulation()

N = 100
yaws = np.zeros(N)
z_os = np.zeros((N, 3))
for simTime in range(N):
    # observe difference from pushing
    px = _observe_block(blockId)
    yaws[simTime] = px[2]
    p.applyExternalForce(blockId, -1, [F, F, 0], [-MAX_ALONG, MAX_ALONG, 0.025], p.LINK_FRAME)
    p.stepSimulation()
    while not _static_environment():
        for _ in range(100):
            p.stepSimulation()
    cx = _observe_block(blockId)
    # difference in world frame
    dx = get_dx(px, cx)
    dz = dx_to_dz(px, dx)
    z_os[simTime] = dz
    logger.info("dx %s dz %s", dx, dz)
    time.sleep(0.1)
logger.info(z_os.std(0) / np.abs(np.mean(z_os, 0)))
plt.scatter(yaws, z_os[:, 2])
plt.xlabel('yaw')
plt.ylabel('dyaw')
plt.show()
```
URDF to be saved as "block_big.urdf" in the same directory:
```
<robot name="block_big" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <link name="base_link">
    <contact>
      <lateral_friction value="0.9"/>
      <spinning_friction value=".1"/>
      <rolling_friction value=".1"/>
    </contact>
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <mass value="6.0"/>
      <inertia ixx="0.1525" ixy="0" ixz="0" iyy="0.1525" iyz="0" izz="0.3025"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
      <geometry>
        <box size="0.55 0.55 0.05"/>
      </geometry>
      <material name="blockmat">
        <color rgba="0.1 0.7 0.1 0.8"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
      <geometry>
        <box size="0.55 0.55 0.05"/>
      </geometry>
    </collision>
  </link>
</robot>
```

(posted to pybullet forums, but maybe I'll get more luck here)

rdyro, 2020-01-30:
I'm trying out a rear wheel drive racecar simulation and the lateral wheel friction seems to be computed by clipping along each axis instead of the norm. My car snaps to simulation axes when drifting.

Here's a minimal example that takes keyboard input. Check it out.

```python
import pybullet as p, pybullet_data, time, sys, math

# configure the engine
p.connect(p.GUI, options="--width=1000 --height=800")
p.setAdditionalSearchPath(pybullet_data.getDataPath())
p.resetSimulation()
p.setGravity(0, 0, -10)
p.setPhysicsEngineParameter(enableConeFriction=1) # ensure cone friction
useRealTime = 0
p.setRealTimeSimulation(useRealTime)

# load objects
plane = p.loadURDF("plane.urdf")
car = p.loadURDF("racecar/racecar.urdf")
front_wheels, rear_wheels, steering = [5, 7], [2, 3], [4, 6]

# change friction coefficients
friction_kwargs = {
        "lateralFriction": 0.5, 
        #"rollingFriction": 0, 
        #"spinningFriction": 0,
        }
p.changeDynamics(plane, -1, **friction_kwargs)
for wheel in front_wheels + rear_wheels:
    p.changeDynamics(car, wheel, **friction_kwargs)

# run the interactive sim
FPS = 60.0
STRSPD, GASTRQ, BRKTRQ = 5.0, 2e0, 1e0
sign = lambda x: x / abs(x) if x != 0 else 0
t_frame = time.time()
while (True):
    # handle user input
    keys = p.getKeyboardEvents()
    steer, gas, brake = 0.0, 0.0, 0.0
    if p.B3G_LEFT_ARROW in keys:
        if keys[p.B3G_LEFT_ARROW] == p.KEY_WAS_TRIGGERED:   steer += -STRSPD
        elif keys[p.B3G_LEFT_ARROW] == p.KEY_IS_DOWN:       steer += -STRSPD
    if p.B3G_RIGHT_ARROW in keys:
        if keys[p.B3G_RIGHT_ARROW] == p.KEY_WAS_TRIGGERED:  steer += STRPSD
        elif keys[p.B3G_RIGHT_ARROW] == p.KEY_IS_DOWN:      steer += STRSPD
    if p.B3G_UP_ARROW in keys:
        if keys[p.B3G_UP_ARROW] == p.KEY_WAS_TRIGGERED:     gas += GASTRQ
        elif keys[p.B3G_UP_ARROW] == p.KEY_IS_DOWN:         gas += GASTRQ
    if p.B3G_DOWN_ARROW in keys:
        if keys[p.B3G_DOWN_ARROW] == p.KEY_WAS_TRIGGERED:   brake += BRKTRQ
        elif keys[p.B3G_DOWN_ARROW] == p.KEY_IS_DOWN:       brake += BRKTRQ
    if ord("q") in keys: break

    # accelerate using a rear wheel drive
    for wheel in rear_wheels:
        p.setJointMotorControl2(car, wheel, p.TORQUE_CONTROL, force=gas)

    # brake with all wheels
    for wheel in rear_wheels + front_wheels:
        p.setJointMotorControl2(car, wheel, p.VELOCITY_CONTROL, 
                targetVelocity=0, force=brake)

    # steer 
    for joint in steering:
        p.setJointMotorControl2(car, joint, p.POSITION_CONTROL, 
                targetPosition=-sign(steer) * 0.4)

    # step the simulation
    if useRealTime == 0: 
        for i in range(round(240.0 / FPS)): 
            p.stepSimulation()

    # adjust the camera
    state = p.getLinkState(car, 0)
    pos, orient = state[0], p.getEulerFromQuaternion(state[1])[2]
    p.resetDebugVisualizerCamera(2, orient / math.pi * 180 - 90, -40, pos)

    # keep FPS
    delay = 1.0 / FPS - (time.time() - t_frame)
    if useRealTime == 0 and delay > 0: time.sleep(delay)
    t_frame = time.time()
```

This might be related to [https://github.com/bulletphysics/bullet3/issues/2117](url)

LemonPi, 2020-01-30:
Thanks for pointing this out! Seems like the fix was patched in a month ago; so I should probably just wait for new release or clone and install locally? I'll post if this patch fixes this issue.

rdyro, 2020-01-30:
I just built master and the issue is still there.

zren96, 2020-04-13:
I encountered a similar issue when applying external force to a grasped object. I simulated the exact same setting except for implicit cone or pyramid cone friction, and somehow the object can slip with default implicit cone but not with pyramid cone. This does not make sense to me since I assume pyramid cone is always smaller. And object can slip in a very weird trajectory.

closed 2021-04-26 by erwincoumans