bulletphysics/bullet3 #2591: loadurdf baseposition change

Hayleymol, 2020-01-22:
Hi, I am trying to train laikago to talk in openai gym environment by fixing codes provided by pybullet.
While I am loading laikago.urdf using loadURDF function, I got an issue. 

When I assign basePosition and baseOrientation using optional inputs like,

self._p.loadURDF(os.path.join("/home/hayley/.mujoco/stable-baseline/laikago/", self.model_urdf),
                                     basePosition=self.basePosition,
                                     baseOrientation=self.baseOrientation,
                                     useFixedBase=self.fixed_base,
                                     flags=pybullet.URDF_USE_SELF_COLLISION))

My laikago does appear in the environment. 
However, when I start training the laikago, it seems like basePosition and baseOrientation is fixed to the world coordinate that it's body won't move even though legs are moving. 
As result,  laikago looks like it is walking in the air.

Loading script that I brought in above is under robot_bases.py script
MJCFBasedRobot class does not have input of basePosition and baseOrientation when loading character. and I guess that is why other xml based characters such as ant baseposition is not fixed during simulation.

I need to fix position and orientation just once when first loading the model.
I try function such as resetBasePositionAndOrientation and tried placing functions in different places but I am having no luck.

Is there way that I can first fix the position and orientation just once when loading?
Where should I place the code in order to initiate only once? 

All helps will be greatly appreciated.

felixxfu, 2020-01-22:
What actions are you applying to the Laikago robot, when you said that "laikago looks like it is walking in the air"?

Maybe try applying random actions and see if it would move?

Hayleymol, 2020-01-24:
I found the issue, turned out that the alive bonus I set was wrong

closed 2020-01-24 by Hayleymol