bulletphysics/bullet3 #2595: Problem with libEGL since last update

guhur, 2020-01-23:
Ubuntu just updated mesa and now PyBullet is not able anymore to create a GL context. I got such error codes:

```
GetDisplay 0 failed with error: 3000
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
GetDisplay 1 failed with error: 3000
GetDisplay 1 failed with error: 3000
GetDisplay 1 failed with error: 3000
libEGL warning: DRI2: failed to create dri screen
GetDisplay 1 failed with error: 3000
GetDisplay 2 failed with error: 3000
GetDisplay 2 failed with error: 3000
Loaded EGL 1.4 after reload.
Loaded EGL 1.4 after reload.
GetDisplay 2 failed with error: 3000
Loaded EGL 1.4 after reload.
GetDisplay 2 failed with error: 3000
Loaded EGL 1.4 after reload.
GL_VENDOR=VMware, Inc.
GL_RENDERER=llvmpipe (LLVM 9.0, 256 bits)
GL_VERSION=3.1 Mesa 19.2.1
GL_SHADING_LANGUAGE_VERSION=1.40
Version = 3.1 Mesa 19.2.1
Vendor = VMware, Inc.
Renderer = llvmpipe (LLVM 9.0, 256 bits)
GL_VENDOR=VMware, Inc.
GL_RENDERER=llvmpipe (LLVM 9.0, 256 bits)
GL_VERSION=3.1 Mesa 19.2.1
GL_SHADING_LANGUAGE_VERSION=1.40
Version = 3.1 Mesa 19.2.1
Vendor = VMware, Inc.
Renderer = llvmpipe (LLVM 9.0, 256 bits)
GL_VENDOR=VMware, Inc.
GL_RENDERER=llvmpipe (LLVM 9.0, 256 bits)
GL_VERSION=3.1 Mesa 19.2.1
GL_SHADING_LANGUAGE_VERSION=1.40
Version = 3.1 Mesa 19.2.1
Vendor = VMware, Inc.
Renderer = llvmpipe (LLVM 9.0, 256 bits)
GL_VENDOR=VMware, Inc.
GL_RENDERER=llvmpipe (LLVM 9.0, 256 bits)
GL_VERSION=3.1 Mesa 19.2.1
GL_SHADING_LANGUAGE_VERSION=1.40
Version = 3.1 Mesa 19.2.1
Vendor = VMware, Inc.
Renderer = llvmpipe (LLVM 9.0, 256 bits)
Compile failed:
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES
```

I am particularly surprised to not seeing my GPU, as it used to.
I am not sure this is exactly connected to PyBullet, but since it's the only software that complains, I guess you might have some insights about it.

erwincoumans, 2020-01-24:
It must be some Ubuntu/egl issue, can't support those. Good luck!

closed 2020-01-24 by erwincoumans

initmaks, 2020-02-16:
Hey @guhur, I am facing the same issue with my setup. Were you able to resolve it?

dannyliu15, 2020-02-19:
I've tried to add the [flag](https://unix.stackexchange.com/a/451214) before running the program, and it works~!

guhur, 2020-02-19:
Great that works with me too:

```
export MESA_GL_VERSION_OVERRIDE=3.3 
export MESA_GLSL_VERSION_OVERRIDE=330
```

erwincoumans, 2020-03-17:
This also works in Colab with hardware EGL:
https://colab.research.google.com/drive/1u6j7JOqM05vUUjpVp5VNk0pd8q-vqGlx

~~~~~~~~~
import os
os.environ['MESA_GL_VERSION_OVERRIDE'] = '3.3'
os.environ['MESA_GLSL_VERSION_OVERRIDE'] = '330'
~~~~~~~~~