bulletphysics/bullet3 #2601: obj2sdf weird rendering

Pandhariix, 2020-01-30:
Hi,

I recently rebased my fork of bullet3 on master and recompiled everything. I used obj2sdf to decompose existing obj files (torso and base of the robot, see pictures), and I get the following result:
[image]

The shapes of the meshes are accurate, but the textures are wrong. I eventually managed to have accurate textures, but at the cost of the meshes' smoothness (by exporting the meshes from blender as objs, without the normals).

I noticed that obj2sdf was updated in PR #2366, so I checked out an older version of master (c7fa8bfc58ac9cb0e7881c880ede86d3ef333bc4), recompiled everything and re-used obj2sdf on my obj files:
[image]

With the older version, everything seems to work

erwincoumans, 2020-01-30:
Thanks for the report. It must have happened when we updated the tiny obj loader.
Do you have a link to obj file + mtl + textures to test?

fuchuyuan, 2020-01-31:
@Pandhariix Can you try to patch [this](https://github.com/bulletphysics/bullet3/pull/2602) and see if this fixes? If not can you send a link for the obj files and texture files for me to test?

Pandhariix, 2020-01-31:
@erwincoumans, unfortunately I cannot yet share those meshes (due to licensing issues), I'll try @fuchuyuan 's fix locally and report the result here though

Pandhariix, 2020-01-31:
@fuchuyuan: it works :+1: 
[image]

closed 2020-02-02 by Pandhariix