bulletphysics/bullet3 #2603: Exposing forward kinematics and forward dynamics function

soshishimada, 2020-02-01:
Would it be possible to expose forward kinematics and forward dynamics function?
Something like, 

jointCoordinates = pybullet.forwardKinematics(bodyUniqueId, jointAngles)
jointAccelerations = pybullet.forwardDynamics(bodyUniqueId, UserInput, externalForce)

It would be quite helpful especially for research use, e.g. implementing a custom inverse kinematics function.

seann999, 2021-02-20:
Bump; the documentation mentions forward kinematics in the table of contents but there is no such function.

erwincoumans, 2021-02-20:
There is: getJointState and getLinkState(s) with the compute forward dynamics flag enabled.

closed 2021-02-20 by erwincoumans

logan-dunbar, 2021-03-01:
@erwincoumans I saw this functionality, but doesn't it require the robot to already be in the specific configuration? How would I get the forward kinematics of an arbitrary configuration of jointAngles as per the original question (without setting the robot to the config)?

erwincoumans, 2021-03-01:
The easiest for now is create a second DIRECT pybullet connection, load the arm there, reset it to the angles, and run the forward kinematics in that second 'dummy' robot.

logan-dunbar, 2021-03-02:
Thanks @erwincoumans, thought as much, probably more efficient to just query the urdf and do the computation by hand, but for now I'm lazy, so I'll use what I've got!  Thanks for the library btw, documentation and examples are a bit thin, but it's way faster than Gazebo, and I will deal with quite a few transition issues for that benefit alone! :)

lorepieri8, 2022-11-24:
+1 to see this feature live.