bulletphysics/bullet3 #2606: URDF limits on prismatic joint do not work

themightyoarfish, 2020-02-04:
I have a fairly trivial joint like this:

```
  <joint name="trolley_to_hoist" type="prismatic">
    <axis xyz="0 0 1"/>
    <origin xyz="0 0 0"/>
    <parent link="trolley"/>
    <child link="hoist"/>
    <limit lower="0.001" upper="0.001" velocity="1"/>
  </joint>
```

According to everything I read, limits on prismatic joints are supported. However, when using `setJointMotorControl()` on the joint, I have no problem moving the joint outside of its limits.

What is the proper way to enable joint limits in URDF that will be respected by Bullet? Or is this not possible?

themightyoarfish, 2020-02-04:
The same holds for revolute joints by the way.

erwincoumans, 2020-03-26:
lower limit needs to be smaller than upper limit, and motor not too strong.

use a fixed joint instead, if lower == upper limit.

closed 2020-03-26 by erwincoumans

goodnamegood, 2020-04-15:
@themightyoarfish are you able to solve this problem of respecting the limits? What is the way to choose the force to be applied?