bulletphysics/bullet3 #2610: pybullet: shadows working only with GUI_on

breznak, 2020-02-06:
### Setup: 

Linux, pybullet 2.5.3, also with latest 2.6; python 3.7.5

### Problem 

In my environment I have 

`GUI_on` handling: 
```
# Set GUI
if self._GUI_on:
  cid = p.connect(p.SHARED_MEMORY)
  if (cid < 0):
    cid = p.connect(p.GUI)
  p.resetDebugVisualizerCamera(3.3, 180, -41, [0.52, -0.2, -0.33]) # Set GUI camera
else:
  p.connect(p.DIRECT)
```

And then I want shadows enabled: 
```
p.configureDebugVisualizer(p.COV_ENABLE_SHADOWS, 1) # Shadows on/off
p.configureDebugVisualizer(p.COV_ENABLE_GUI,0) # GUI off
```

### Expected behavior: 

shadows shown always (regardless of GUI_on variable)

### Actual behavior

With `GUI_on=True`, shawdows are shown,
while with `GUI_on=False` the shadows are not rendered. 

Is this a bug or something incorrect in my setting? Thank you,

shpigi, 2021-01-04:
@breznak , were you able to solve this? Having same issue

breznak, 2021-01-06:
@shpigi unfortunately I still haven't solved that. I'm not dealing with that problem anymore, though

closed 2021-04-26 by erwincoumans