bulletphysics/bullet3 #2614: Any API for adjust collision margin in pybullet ?

v-wewei, 2020-02-07:
Since Collision margin play a quite important variable for assemble and grasp task, so would you like to add such a api for us for convinient. @erikogenvik 
Much Thanks for your effort for such a great physic simulator . Hope this suggestion will be accept.

erwincoumans, 2020-02-10:
Yes, it can be exposed. See also https://github.com/bulletphysics/bullet3/issues/2584
Will do it soon, but note that you will have to resolve any issues due to changing the default collision margins.

erwincoumans, 2020-02-15:
See https://github.com/bulletphysics/bullet3/pull/2626

v-wewei, 2020-02-15:
> See #2626 
Does it support now ? i see the tag is unsupported. @erwincoumans 
seems the c++ version support the option for collision margin, right?

v-wewei, 2020-03-02:
@erwincoumans 
I am not able to adjust the collision margin for the gripper of my yumi-robot, seems the API can only change the collision margin for basic shape like cylinder and sphere and so on. Would it be able to adjust collision margin for any other convex hull just like the collision in blender. Thanks very much!

erwincoumans, 2020-03-02:
Can you share a reproduction case with actual urdf and meshes (all in a zip file)?

v-wewei, 2020-03-02:
@erwincoumans the file is more than 10 MB, how can i show it to you ？

v-wewei, 2020-03-02:
would you like to provide your email address for convinience , that i can send you through the eamil . the total zip file size is about 40MB large.

closed 2021-04-26 by erwincoumans