bulletphysics/bullet3 #2615: Missing mesh file in minitaur_terrain_randomizer

Visgean, 2020-02-12:
Hi, this file 
https://github.com/bulletphysics/bullet3/blob/aec9968e281faca7bc56bc05ccaf0ef29d82d062/examples/pybullet/gym/pybullet_envs/minitaur/envs/env_randomizers/minitaur_terrain_randomizer.py#L215

references mesh file "robotics/reinforcement_learning/minitaur/envs/testdata/triangle_mesh_terrain/terrain9735.obj"
but this is nowhere to be found...

erwincoumans, 2020-02-12:
Here is the file, it isn't very interesting:
[terrain9735.obj.zip](https://github.com/bulletphysics/bullet3/files/4189723/terrain9735.obj.zip)

closed 2020-02-12 by erwincoumans

Visgean, 2020-02-12:
Thanks! Do you happen to have the code that generated the mesh? I was planning to write something similar myself but if it has already been written.... I am planning to use this sort of randomized environment for my dissertation.