bulletphysics/bullet3 #2628: Get orthographic projection camera

LemonPi, 2020-02-16:
I'd like to have an orthographic view of my simulation (basically top down 2D). Currently, I think it's possible to get an orthographic image of the scene with `getCameraImage` since that takes in `viewMatrix` and `projectionMatrix`, but we're stuck with a projective camera for debugging. Can we allow the debug view/video view to also allow for custom camera projections?

erwincoumans, 2020-03-01:
It is not supported at the moment. Even in getCameraImage, we compute near/far from the projection matrix, which doesn't work for prthogonal matrices.
We could modify both getCameraImage and resetDebugVisualizerCamera to support orthogonal matrices. Contributions welcome.

erwincoumans, 2020-03-01:
(closing for now, will re-open if a contribution pull request shows up)

closed 2020-03-01 by erwincoumans