bulletphysics/bullet3 #2636: bullet 2.89 source compilation trap: debug64 and release64 both build with BT_USE_DOUBLE_PRECISION enabled, but the headers default to SINGLE precision.

etc0de, 2020-02-19:
There is quite the compilation trap when trying to build bullet physics 2.89 from source right now:

The debug64 and release64 premake targets both build with BT_USE_DOUBLE_PRECISION enabled, but the headers by default seem to assume its off. Therefore, things will blow up with super weird valgrind errors like this one:

```
==183956== Invalid write of size 1
==183956==    at 0x48FD0A: btAlignedObjectArray<btAlignedObjectArray<btDbvtNode const*> >::init() (btAlignedObjectArray.h:86)
==183956==    by 0x48F783: btAlignedObjectArray<btAlignedObjectArray<btDbvtNode const*> >::btAlignedObjectArray() (btAlignedObjectArray.h:123)
==183956==    by 0x48B8CC: btDbvtBroadphase::btDbvtBroadphase(btOverlappingPairCache*) (btDbvtBroadphase.cpp:131)
==183956==    by 0x453EBC: colworld_NewWorld (colworld.cpp:73)
[... more backtrace in my app code ...]
==183956==  Address 0x4e388d0 is 0 bytes after a block of size 256 alloc'd
==183956==    at 0x4839E86: operator new(unsigned long) (vg_replace_malloc.c:344)
==183956==    by 0x453EAC: colworld_NewWorld (colworld.cpp:73)
[... more backtrace in my app code ...]
```
... when building with trivial default settings like this:
```
cd build3/ && ./premake4_linux64 --double gmake && cd gmake && make config=debug64 BulletCollision LinearMath BulletDynamics
```
And doing something simple like this:
```
#include <btBulletDynamicsCommon.h>
...
 btBroadphaseInterface *broadphase = new btDbvtBroadphase(); 
```

As a result I debugged around the constructor for an hour and was about to file a ticket until the related search showed me #1330 that made me realize what is going on. I think I am probably not the only one who will run into this, so I suggest it should be changed.

**Suggestion:** alter the headers to already define `BT_USE_DOUBLE_PRECISION` by default with something like `#define BT_USE_SINGLE_PRECISION` for the user to turn it off, or alter the build defaults of the premake script are changed to build with single precision. This will fix that source + headers in unaltered combination are set to blow up with these differing defaults, which I think is somewhat unfortunate as a first experience.

etc0de, 2020-02-24:
I actually had used `--double` on gmake without realizing, so this is actually my fault :woman_facepalming:

closed 2020-02-24 by etc0de

erwincoumans, 2020-02-24:
It is tricky indeed. The next phydics engine is header only so it won't happen.