bulletphysics/bullet3 #2639: Different behavior for the first reset of envrionment

johannespitz, 2020-02-20:
https://github.com/bulletphysics/bullet3/blob/cbede4eb6c4b9b6d0dddf5b3964a1c7449af05d6/examples/pybullet/gym/pybullet_envs/gym_locomotion_envs.py#L27

Is there a reason for this `stateId` distinction?
I believe this leads to weird behavior when resetting gym environments like `AntBulletEnv-v0`.

```
>>> env = gym.make('AntBulletEnv-v0')
>>> env.seed(9)
[9]
>>> env.reset()
array([ 0.0000000e+00,  **6.0072844e-06**,  1.0000000e+00,  0.0000000e+00,
        0.0000000e+00,  0.0000000e+00,  0.0000000e+00, -0.0000000e+00,
       -1.5792921e-02,  0.0000000e+00, -1.7883309e+00,  0.0000000e+00,
        2.5327675e-02,  0.0000000e+00,  1.9404743e+00,  0.0000000e+00,
        1.2966397e-01,  0.0000000e+00,  1.8370057e+00,  0.0000000e+00,
       -9.5518544e-02,  0.0000000e+00, -1.8426850e+00,  0.0000000e+00,
        0.0000000e+00,  0.0000000e+00,  0.0000000e+00,  0.0000000e+00],
      dtype=float32)
>>> env.seed(9)
[9]
>>> env.reset()
array([ 0.0000000e+00,  **5.7342259e-06**,  1.0000000e+00,  0.0000000e+00,
        0.0000000e+00,  0.0000000e+00,  0.0000000e+00, -0.0000000e+00,
       -1.5792921e-02,  0.0000000e+00, -1.7883309e+00,  0.0000000e+00,
        2.5327675e-02,  0.0000000e+00,  1.9404743e+00,  0.0000000e+00,
        1.2966397e-01,  0.0000000e+00,  1.8370057e+00,  0.0000000e+00,
       -9.5518544e-02,  0.0000000e+00, -1.8426850e+00,  0.0000000e+00,
        0.0000000e+00,  0.0000000e+00,  0.0000000e+00,  0.0000000e+00],
      dtype=float32)
>>> env.seed(9)
[9]
>>> env.reset()
array([ 0.0000000e+00,  **5.7342259e-06**,  1.0000000e+00,  0.0000000e+00,
        0.0000000e+00,  0.0000000e+00,  0.0000000e+00, -0.0000000e+00,
       -1.5792921e-02,  0.0000000e+00, -1.7883309e+00,  0.0000000e+00,
        2.5327675e-02,  0.0000000e+00,  1.9404743e+00,  0.0000000e+00,
        1.2966397e-01,  0.0000000e+00,  1.8370057e+00,  0.0000000e+00,
       -9.5518544e-02,  0.0000000e+00, -1.8426850e+00,  0.0000000e+00,
        0.0000000e+00,  0.0000000e+00,  0.0000000e+00,  0.0000000e+00],
      dtype=float32)
```

erwincoumans, 2020-02-29:
The self.parts seem to be randomized somewhere, @benelot  any idea why the order is not fixed?
We need to reverse engineer this code to track down what causes the random order of self.parts

The self.stateId >= 0 is unrelated to the randomness of this environment (just comment out the save/restoreState to check it).

erwincoumans, 2020-03-01:
I tracked it down, there are a few issues:
1) we postpone the robot/environment creation until the first reset.
if you call 'render(mode='human') before the first reset, you will get a GUI window.
Since this first reset is doing more work, it makes mode calls to random, hence different results.

Probably not going to change this yet, since it changes behavior and previous learning results.

Here is a patch to make all reset's produce the same results for those Gym environments.
[patch.zip](https://github.com/bulletphysics/bullet3/files/4271233/patch.zip)

erwincoumans, 2020-03-01:
Please use the patch if it is important for you. I'm not going to break compatibility for v0 envs, will consider it for future new environments!

closed 2020-03-01 by erwincoumans