bulletphysics/bullet3 #2646: Contact normals persisting even after rotation.

kermado, 2020-02-27:
I have a simple scene that consists of a box resting on a platform. Both objects have a single box shape. The platform is kinematic whilst the box object is dynamic. The platform rotates about its center. The expectation is for the contact normals to rotate as the platform rotates. However, this doesn't happen.

[image]

It seems that dBoxBox2 doesn't report any contacts after the first time-step. I guess that the contacts generated in the first time-step are persisted and the impulses are sufficient to prevent the generation of penetrating contacts for all other time-steps.

Shouldn't dBoxBox2 account for the contact breaking threshold in the separating axis tests? Specifically, it feels like the TST macros should be checking whether the separation is greater than the contact breaking threshold before returning early. The same for counting the number of penetrating contacts also.

The relevant lines are [330](https://github.com/bulletphysics/bullet3/blob/cdd56e46411527772711da5357c856a90ad9ea67/src/BulletCollision/CollisionDispatch/btBoxBoxDetector.cpp#L330), [358](https://github.com/bulletphysics/bullet3/blob/cdd56e46411527772711da5357c856a90ad9ea67/src/BulletCollision/CollisionDispatch/btBoxBoxDetector.cpp#L358) and [648](https://github.com/bulletphysics/bullet3/blob/cdd56e46411527772711da5357c856a90ad9ea67/src/BulletCollision/CollisionDispatch/btBoxBoxDetector.cpp#L648)

After these changes, I get the behaviour that I was expecting:

[image]

I guess that the other collision tests will be doing pretty much the same.

erwincoumans, 2020-02-27:
This is likely due to contact caching in btPersistentContactManifold.
We could/should add a check (dot product) of new normal against the old contact normal (both are in world space) and if they are too different we need to remove the old normal. Or better, always replace the contact normal. Need to check this. Do you have a simple Bullet reproduction case?

kermado, 2020-02-28:
That was my initial suspicion, but after following it through with the debugger I saw that, except for the first frame or so, no contact points were even being reported by `btBoxBoxDetector::getClosestPoints()`, which calls `dBoxBox2()`. It was always exiting early due to one of the separating axis tests. If we don't get any contact points and normals, then I don't see how the contact caching is to blame.

Also, `replaceContactPoint()` appears to replace the contact normal regardless, although I can see that this function has changed from my older version of Bullet. The version in the trunk appears to try to keep the contact point if the friction force is within the friction cone. My older function looks something like this:

```
void replaceContactPoint(const btManifoldPoint& newPoint, int insertIndex)
{
	int lifeTime = m_pointCache[insertIndex].getLifeTime();
	btScalar appliedImpulse = m_pointCache[insertIndex].m_appliedImpulse;
	btScalar appliedLateralImpulse1 = m_pointCache[insertIndex].m_appliedImpulseLateral1;
	btScalar appliedLateralImpulse2 = m_pointCache[insertIndex].m_appliedImpulseLateral2;

	void* cache = m_pointCache[insertIndex].m_userPersistentData;
		
	m_pointCache[insertIndex] = newPoint;
	m_pointCache[insertIndex].m_userPersistentData = cache;
	m_pointCache[insertIndex].m_appliedImpulse = appliedImpulse;
	m_pointCache[insertIndex].m_appliedImpulseLateral1 = appliedLateralImpulse1;
	m_pointCache[insertIndex].m_appliedImpulseLateral2 = appliedLateralImpulse2;

	m_pointCache[insertIndex].m_lifeTime = lifeTime;
}
```

I'll try create a simple test case later today.

kermado, 2020-02-28:
To follow up, I haven't been able to reproduce this in the latest version of Bullet. I'm going to close this issue until I'm able to reproduce. Thanks.

closed 2020-02-28 by kermado